gqrx-sdr/gqrx #1118: FFT/waterfall: proper amplitude averaging

jenda122, 2022-04-17:
Currently, the spectrum/waterfall display works in a way that the GUI calls `get_fft_data` on a timer, which makes one transform at that exact time point. When receiving a pulsed signal (e.g. packet data transmission or pulse radar), this leads to lost transmissions and/or time aliasing of the pulses. One workaround is to set a really huge FFT size and rate to have some overlap, however, the largest FFTs are too intensive to compute and it starts dropping the signal too.

I think the correct solution is to transform the samples as they are being received in [rx_fft_c::work](https://github.com/gqrx-sdr/gqrx/blob/master/src/dsp/rx_fft.cpp#L81), accumulate the results in a temporary buffer and when the GUI asks for spectrum data, return this buffer as the average.

Additional feature would be to compute maximum instead of average, which may be interesting when looking for sparse short transmissions.

Should I try to implement this, or would you recommend some other approach?

argilo, 2022-04-17:
> [...] the largest FFTs are too intensive to compute and it starts dropping the signal too.
>
> I think the correct solution is to transform the samples as they are being received in [...]

I suspect you might run into the same problem with smaller but more frequent FFTs. One way or another, you'd need to handle the situation where you can't keep up with the incoming sample stream.

> Should I try to implement this

You're certainly welcome to experiment. Of course, I can't guarantee that any particular change will be accepted into an upcoming release.

vladisslav2011, 2022-04-24:
Hello.

> One workaround is to set a really huge FFT size and rate to have some overlap

That's correct workaround. You may decrease input sample rate to make FFTs overlap at lower frame rate, that your machine would be able to handle.

> I think the correct solution is to transform the samples as they are being received in [rx_fft_c::work](https://github.com/gqrx-sdr/gqrx/blob/master/src/dsp/rx_fft.cpp#L81), accumulate the results in a temporary buffer and when the GUI asks for spectrum data, return this buffer as the average.

In this case if your machine is not powerful enough to handle longer overlapped FFTs, you'll have dropouts too. And you'll have choppy sound/increasing tuning lag as GNU Radio would try to grow `rx_fft`'s input buffer as it's `work` would not return fast enough.

As alternative to moving the FFT calculation to `rx_fft::work`, you may split a window application and actual FFT calculation to different threads and, if that's not enough, you may implement multithreaded FFT calculation/rendering. This would help in case your machine have low single thread performance, but have many enough cores.