gqrx-sdr/gqrx #1119: GQRX hangs after working for some time

alessss, 2022-04-20:
After running GQRX hangs and becomes unresponsive. Here is some info about hardware. Let me know if some addiional problem info needed.

`
Model: MacBookPro15,1, BootROM 1715.60.4.0.0 (iBridge: 19.16.10631.5.3,0), 6 processors, 6-Core Intel Core i7, 2,6 GHz, 32 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Display: Color LCD, 2880 x 1800 Retina, Main, MirrorOff, Online
Graphics: Radeon Pro 560X, Radeon Pro 560X, PCIe, 4 GB
Memory Module: BANK 0/ChannelA-DIMM0, 16 GB, DDR4, 2400 MHz, SK Hynix, H5ANAG8NAMR-UHC
Memory Module: BANK 2/ChannelB-DIMM0, 16 GB, DDR4, 2400 MHz, SK Hynix, H5ANAG8NAMR-UHC
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Jul 12 2021 19:26:30 version 9.30.464.0.32.5.76 FWID 01-45ccefcd
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB3.0 Hub
USB Device: Flash Card Reader/Writer
USB Device: USB31Bus
USB Device: USB2.0 Hub
USB Device: RTL2838UHIDIR
USB Device: USB 2.0 BILLBOARD
USB Device: USB Billboard Device
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5`

vladisslav2011, 2022-04-24:
Hello.
Thanks for a bug report.

Could you...
1. Run Gqrx under gdb, break it with Ctrl+C when it becomes unresponsive, obtain backtrace with `bt` command and post it here.
2. Run Gqrx from console. Run `top` in a separate console with detailed thread information enabled ("H", "V", I guess). Wait until Gqrx becomes unresponsive. Post Gqrx threads CPU usage information from `top` here (scroll it with arrows, then make a screenshot).

argilo, 2023-05-07:
Unfortunately there is not enough information in this report for me to do anything with it, so I'll close it off.

closed 2023-05-07 by argilo