gqrx-sdr/gqrx #1120: Crash on startup with SIGSEGV in gr::funcube::fcdpp::make()

rundel-tech, 2022-04-23:
After installing Ubuntu 22.04 LTS and installing gqrx ("gqrx-sdr") from the repositories, gqrx crashes consistently on startup. It worked fine on 21.10. The SDR is a Funcube Pro+. Starting from the console gives:

merlin@thebe:~$ gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
file_source :warning: file size is not a multiple of item size
Resampling audio 96000 -> 48000
BandPlanFile is /home/merlin/.config/gqrx/bandplan.csv
BookmarksFile is /home/merlin/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
libusb: warning [libusb_exit] device 4.1 still referenced
libusb: warning [libusb_exit] device 3.1 still referenced
libusb: warning [libusb_exit] device 2.1 still referenced
libusb: warning [libusb_exit] device 1.9 still referenced
libusb: warning [libusb_exit] device 1.7 still referenced
libusb: warning [libusb_exit] device 1.11 still referenced
libusb: warning [libusb_exit] device 1.10 still referenced
libusb: warning [libusb_exit] device 1.8 still referenced
libusb: warning [libusb_exit] device 1.6 still referenced
libusb: warning [libusb_exit] device 1.4 still referenced
libusb: warning [libusb_exit] device 1.3 still referenced
libusb: warning [libusb_exit] device 1.5 still referenced
libusb: warning [libusb_exit] device 1.2 still referenced
libusb: warning [libusb_exit] device 1.1 still referenced
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using FUNcube Dongle V2.0 (hw:1)
Segmentation fault (core dumped)

From the crash report:
SegvAnalysis:
 Segfault happened at: 0x7f9f8f273ab5:	mov    0x20(%rax),%rbp
 PC (0x7f9f8f273ab5) ok
 source "0x20(%rax)" (0x00000020) not located in a known VMA region (needed readable region)!
 destination "%rbp" ok
SegvReason: reading NULL VMA
SourcePackage: gqrx-sdr

argilo, 2022-04-24:
The crash occurs in gr-funcube. I downloaded source and debug symbols to have a look at the crash:
```
Thread 1 "gqrx" received signal SIGSEGV, Segmentation fault.
gr::funcube::fcdpp_impl::fcdpp_impl (this=<optimized out>, user_device_name=..., unit=<optimized out>, this=<optimized out>, user_device_name=..., unit=<optimized out>) at ./lib/fcdpp_impl.cc:86
warning: Source file is more recent than executable.
86	        GR_LOG_INFO(d_logger, boost::format("Audio device %1% opened") % device_name);
(gdb) bt
#0  gr::funcube::fcdpp_impl::fcdpp_impl (this=<optimized out>, user_device_name=..., unit=<optimized out>, this=<optimized out>, 
    user_device_name=..., unit=<optimized out>) at ./lib/fcdpp_impl.cc:86
#1  0x00007ffff59c7ac5 in gr::funcube::fcdpp::make (device_name=..., unit=1) at ./lib/fcdpp_impl.cc:25
#2  0x00007ffff7d27ba5 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-osmosdr.so.0.2.0
#3  0x00007ffff7d0acd5 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-osmosdr.so.0.2.0
#4  0x00007ffff7d0e0bb in osmosdr::source::make(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /lib/x86_64-linux-gnu/libgnuradio-osmosdr.so.0.2.0
#5  0x000055555560ff44 in receiver::set_input_device(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
#6  0x00005555555fa3d0 in MainWindow::loadConfig(QString const&, bool, bool) ()
#7  0x0000555555600a12 in MainWindow::MainWindow(QString const&, bool, QWidget*) ()
#8  0x00005555555e41a8 in main ()
```
It looks like it crashes the first time it tries to log something, probably because the logger is not initialized.

Debian applied a patch which removed logger initialization, and I suspect that's the problem:

https://sources.debian.org/patches/gr-funcube/1.0.0-3/update-logging/

rundel-tech, 2022-04-24:
That's interesting. I really appreciate your looking into this. What is the next stage? Should I try to communicate this to the Debian package maintainers, or is the automated crash report sufficient?

argilo, 2022-04-24:
I sent a short message about the problem to the Debian package maintainer, but have not heard back. It probably wouldn't hurt to file a bug report in Ubuntu's bug tracking system if you have the time.

rundel-tech, 2022-04-24:
I have raised this as you suggest: https://bugs.launchpad.net/ubuntu/+source/gqrx-sdr/+bug/1970126