gqrx-sdr/gqrx #1129: Cannot switch between bookmarked channels.

CipherOps, 2022-06-25:
Fedora 35
Repo installed Gqrx 2.15.9
Bookmark issue

I have multiple bookmarks, I cannot switch between them. If I'm on one bookmark, and click another to go to it, it looks like I tuned to it, however the waterfall stays the same, and I'm still on the previous station, it doesn't actually tune. However, if I click from one station to another, and then quit and restart Gqrx, hit play, I'm on the station I switched to, but I cannot switch to bookmarks I have by clicking them.

1. Listening to one bookmarked channel
2. Click another bookmarked channel
3. Interface shows I switched
4. Still on old channel
5. Exit software
6. Relaunch
7. Play
8. I'm on the station I tried to switch to before closing.

If I'm not on a bookmarked station, I can type and change the channel I'm on to another and not have to restart, the issue only exists if I'm on a bookmarked station and try to change.

Rua, 2022-07-03:
I've had an issue like this specifically when the two stations overlap in frequency. It would always choose the lower of the two. Is that what you also experience?

pinkavaj, 2022-11-18:
Can this be related to #1175 or #267 ?

LevitatingBusinessMan, 2023-10-02:
I suddenly have this bug as well.

argilo, 2023-10-02:
Unfortunately I have been unable to reproduce this bug. Perhaps there are some specific conditions that are necessary for it to occur. Any more details you can provide would be much appreciated.

argilo, 2023-10-02:
In particular, a copy of your `~/.config/gqrx/bookmarks.csv` would be helpful.

Also, are you clicking on the bookmark name at the top of the plot, or double-clicking in the bookmarks pane?

LevitatingBusinessMan, 2023-10-02:
Here: https://catbin.xyz/UYJar

I am doubling clicking the bookmark in the pane.

CipherOps, 2023-10-02:
Bookmark pane here as well. Bookmarks can be the same frequency also, if I have it tuned to one, click the bookmark, the UI will show it swapped channels but I still hear the old one.

If I click the bookmark to move channels, exit GQRX, then re-open and hit the "Play" button, I am on the correct channel. 

I've had this across multiple distros with the appimage and system install version. I currently do not have bookmarks setup since it has never worked, but will get around to it later and send the file you have requested.

argilo, 2023-10-02:
Hmm, even with that bookmarks file I'm still not seeing the problem. Could you also share your `~/.config/gqrx/default.conf`, in case the problem depends on particular settings?

LevitatingBusinessMan, 2023-10-02:
https://catbin.xyz/PdufA

argilo, 2023-10-02:
Still no luck reproducing the problem, unfortunately.

Does it happen always, or only sometimes?

CipherOps, 2023-10-02:
In my case it is always, I'll have it open playing a stream, click a bookmark to swap to it, GUI shows it did, but still playing the previous channel, even manually adjusting at that point has no effect.

argilo, 2023-10-02:
And where exactly are you clicking? I do see some odd behaviour if I click on the number column to the left of the frequency column, which I just tried for the first time.

argilo, 2023-10-02:
(Up until this point, I had been double-clicking on the Frequency/Name/Modulation/Bandwidth columns.)

LevitatingBusinessMan, 2023-10-02:
It started happening to my just now but I think I had this bug before. I think I fixed it by using either USB2 or USB3 with my RTLSDR. Because the issue was simply the SDR refusing to respond to the command to change frequency. Somehow it does not happen when setting the frequency manually.

argilo, 2023-10-02:
> I do see some odd behaviour if I click on the number column to the left of the frequency column

Never mind, I confused myself. Clicking on that column doesn't appear to do anything.

argilo, 2023-10-03:
> Because the issue was simply the SDR refusing to respond to the command to change frequency. Somehow it does not happen when setting the frequency manually.

That's an interesting clue. I do see that clicking a bookmark causes `set_center_freq` to be called twice in a row, while adjusting the frequency other ways only results in a single call. Perhaps the extra USB traffic could cause a bug in the RTL-SDR driver to manifest.

CipherOps, 2023-10-03:
That is possible, I too have an RTL-SDR device.

argilo, 2023-10-03:
That wouldn't explain this part though:

> If I'm not on a bookmarked station, I can type and change the channel I'm on to another and not have to restart, the issue only exists if I'm on a bookmarked station and try to change.

So there must be more to it.

Any further clues you can find about when or why this bug occurs would be helpful, since I've still never seen it happen on my machines.

0penBrain, 2023-10-08:
> That's an interesting clue. I do see that clicking a bookmark causes `set_center_freq` to be called twice in a row, while adjusting the frequency other ways only results in a single call.

 1. Yes, because `MainWindow::onBookmarkActivated` directly calls `MainWindow::setNewFrequency`.
 2. This latter directly calls `receiver::set_rf_freq` then calls `CFreqCtrl::setFrequency`.
 3. Again this latter will then emit `newFrequency` signal, that in `MainApplication` is connected to `MainWindow::setNewFrequency`.
 4. Go back step 2

Fortunately this potential loop is protected here : https://github.com/gqrx-sdr/gqrx/blob/154bbe6c62f19e46470a4e9223283a21ddfb5c4f/src/qtgui/freqctrl.cpp#L203-L204

But if you start it (the loop) calling `MainWindow::setNewFrequency`, `set_rf_freq` is called twice in a row.
It's called only once if you start it calling `CFreqCtrl::setFrequency`, so this is what PR #1309 implements.
This said it doesn't goes deeper finding a more complete solution that would totally eliminate the flaw.

argilo, 2023-10-08:
Some other information that might help with diagnosis:

* Which operating system & version are you running?
* Which model of RTL-SDR are you using? Which tuner does it have?

argilo, 2023-10-08:
Also:

* Are you using the RTL-SDR driver package supplied by the operating system? Or something else (e.g. compiled from source, included with Gqrx's AppImage / DMG)?

0penBrain, 2023-10-10:
> * Are you using the RTL-SDR driver package supplied by the operating system?

I'm having a hard time trying to make sure which driver is used. :smile: Do you have any clue how to get this ? Thx

LevitatingBusinessMan, 2023-10-10:
> > * Are you using the RTL-SDR driver package supplied by the operating system?
> 
> I'm having a hard time trying to make sure which driver is used. 😄 Do you have any clue how to get this ? Thx

It would depend on your operating system, what are you on?

0penBrain, 2023-10-11:
> It would depend on your operating system, what are you on?

Oops. 😕 I'm on Linux. Kubuntu 22.04 more precisely.

argilo, 2023-10-11:
On Linux, ``ldd `which gqrx` | grep rtl`` will show the location of the RTL-SDR library you're using, which should be enough to tell what the situation is. I'd be more interested to know this for people who are able to reproduce the bug, though.

mordae, 2024-01-23:
I am on Fedora 39 and I have the same bookmark issue (and more).

In my case, whenever I change the frequency by scrolling at the main frequency input panel at the top of the window, using too large steps results in the RTL-SDR not actually changing the frequency while gqrx thinks it has been changed. It somehow fixes itself when I perform a much smaller increment afterwards.

Same with the bookmarks. As long as the bookmark jump requires tuning to a different frequency, RTL-SDR stays at the original one while gqrx updates the interface. Once I adjust the frequency by 1kHz, RTL-SDR is retuned and everything seems to match.

It was not happening on Fedora 38, I believe.

argilo, 2024-01-23:
Again, this sounds like a hardware or driver issue. Which model of RTL-SDR are you using, and what type of tuner does it have? What version of the RTL-SDR driver are you using?

mordae, 2024-01-27:
I have switched to `rtl-sdr-blog` driver and it seems to work fine now. Sorry to bother you.