gqrx-sdr/gqrx #1131: Problems with signal reception (BladeRF-x40 , GNUradio)

raul8883, 2022-07-01:
Hi everyone, I am having problems when I receive signals using two SDRs (BladeRF - x40), one to transmit and one to receive (I know I can transmit and receive using 1 SDR, but I need the complete schematic transmitter, receiver), the problem is the following, I have a complex signal , with the following values:
in-phase component (i(t)) tx = Ac * x(t)
quadrature component (q(t)) tx = 0
Vy (t) tx = i(t) + j q(t)
TX
[image]

I send this signal to the channel and therefore it is received by the SDR receiver.
What I receive in the reception is not similar to what I transmitted, because of a tone appears both in its component in phase and in quadrature, by making q(t) rx /≈ q(t) tx /≈ 0.
RX
[image]

For a better compression, I will transmit the following signal x(t) = 0 (equivalent to i(t) tx = 0, q(t) tx = 0), so in theory in reception it would only have noise, but this is not the case.
TX
[image]

RX
[image]

When I use a single SDR to transmit and receive this does not happen, I don't know if this is due to some kind of synchronization, but with this problem I can not correctly classify the type of modulation.

BLOCK SDR Transmitter.
[image]

BLOCK SDR Receiver.
[image]

This is what i am getting (Frecuency Domain) when i am transmitting a message with amplitude = 0 and both SDRs have gain equal to 0

[image]

When I increase the gain, that peak increases, in addition another peak appears symmetrical to that one with less amplitude
[image]

This does not happen when I transmit and receive with the same sdr,
Could this be due to mismatches between both SDRs?

I need help to know how to make it not transmit and receive that unwanted signal.
Thank you very much in advance!

vladisslav2011, 2022-07-06:
Hello.
I think, there is better place to ask general GNU Radio questions: GNU Radio matrix chat https://matrix.to/#/#gnuradio:gnuradio.org

argilo, 2023-04-28:
This does not appear to have any connection to Gqrx, so I'll close off the issue.

closed 2023-04-28 by argilo