gqrx-sdr/gqrx #1132: zsh: segmentation fault  gqrx

cx802, 2022-07-02:
┌──(root💀kali)-[~/gqrx.git]
└─# gqrx                                                   
zsh: segmentation fault  gqrx

Linux kali 5.18.0-kali2-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.5-1kali1 (2022-06-20) x86_64 GNU/Linux

how to fix? ;/

vladisslav2011, 2022-07-05:
Hello.
Provide a backtrace, please.
1. Run gqrx under gdb

```
$ gdb gqrx
(gdb) run
```
2. Make it crash
3. Switch to a  terminal with gdb and type `bt`
4. Post backtrace here

I don't think running gqrx under root to be a good idea...

Inglebard, 2022-10-18:
Hi, 
same issue on manjaro.

```
GNU gdb (GDB) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gqrx...
(No debugging symbols found in gqrx)
(gdb) run
Starting program: /usr/bin/gqrx 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7ffff045d6c0 (LWP 21430)]

Thread 1 "gqrx" received signal SIGSEGV, Segmentation fault.
0x00007ffff6f8547c in QFont::operator==(QFont const&) const () from /usr/lib/libQt6Gui.so.6
(gdb) 

```

vladisslav2011, 2022-10-18:
@Inglebard It looks, like there is a problem with either qt6ct theme plugin or missing font. Have you tested other Qt6 applications with qt6ct platform theme?

Inglebard, 2022-10-18:
@vladisslav2011 you are right. I clear my /etc/environment where some QT variable were set. Now it starts properly, thanks.

argilo, 2023-04-16:
I'll close this off, since the problem appears to lie outside Gqrx.

closed 2023-04-16 by argilo