gqrx-sdr/gqrx #1134: AM modulation squelch gets stuck after start

l29ah, 2022-07-08:
Observed on both 2.15.8 and today's master.

Repro:
* set modulation to AM, squelch to -150dB
* restart gqrx
* start capture
* observe no audio, neither in the audio plot, nor in your headphones
[image]

Workaround:
* switch modulation to WFM
* switch back to AM

vladisslav2011, 2022-07-08:
Hello.
There is no bug. You have the AGC disabled and have set gain to 0 db.
You should enable the AGC to listen to AM signals or set the gain manually.

l29ah, 2022-07-09:
AGC is disabled and gain is set to 0dB for WFM as well, yet i hear them perfectly, and after hearing WFM i can hear AM, so there is certainly an inconsistent behavior.

vladisslav2011, 2022-07-31:
I agree.
I have opened a PR to fix this inconsistent behavior https://github.com/gqrx-sdr/gqrx/pull/1096.
This PR (sequence) implements new constant complexity AGC algorithm and enables AGC for all demodulators. It helps when some NFM stations use narrow deviation while others are using wide on the same frequency.

argilo, 2023-04-28:
> Repro:
>
> * set modulation to AM, squelch to -150dB
> * restart gqrx
> * start capture
> * observe no audio, neither in the audio plot, nor in your headphones
>
> Workaround:
>
> * switch modulation to WFM
> * switch back to AM

I was able to reproduce this on the latest version (2.16). In fact, the expected behaviour in this case should be that the audio is very quiet even after switching back from WFM, because AGC is disabled. But it seems like switching from WFM to AM (or other narrow-band modes) causes the AGC to incorrectly switch to "Medium".

argilo, 2023-04-28:
This looks to be a long-standing bug. It's present in Gqrx 2.6 (released in 2016).