gqrx-sdr/gqrx #1137: Crash when changing frequency (QT?)

kevin-nel, 2022-07-26:
Hi I am trying to use gqrx (v 2.15.9) on fedora with an rtl-sdr dongle.

I have got it to detect the sdr and open, when I hit play everything seems fine but as soon as I try to change to frequency (using the square blocks for each digit at the top of the window) the program crashes.

this is the output when I run it from the terminal
``
QSocketNotifier: Can only be used with threads started with QThread
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace airspy soapy redpitaya 
file_source :warning: file size is not a multiple of item size
Resampling audio 96000 -> 48000
BandPlanFile is /home/cheese/.config//gqrx/bandplan.csv
BookmarksFile is /home/cheese/.config//gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 12.0.1 20220129 (Red Hat 12.0.1-0); Boost_107600; UHD_4.1.0.5
Found Rafael Micro R820T tuner
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace airspy soapy redpitaya 
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Found Rafael Micro R820T tuner
[INFO] Using format CF32.
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.
``

It seems like it might be QT related.

vladisslav2011, 2022-07-30:
Hello.

> rtlsdr_demod_write_reg failed with -9
> r82xx_write: i2c wr failed=-9 reg=17 len=1
> r82xx_set_freq: failed=-9

This crash does not look Qt related. It's more likely libusb/librtlsdr related. librtlsdr fails to write R820T registers and throws an exception. Then Qt does not know, what to do with this exception.

Try running gqrx under gdb. If it crashes under gdb too, post gdb output and backtrace here.

skndr, 2022-09-10:
Same issue here

(gdb) run /usr/bin/gqrx
Starting program:  /usr/bin/gqrx
No executable file specified.
Use the "file" or "exec-file" command.
(gdb) file /usr/bin/gqrx
Reading symbols from /usr/bin/gqrx...
(No debugging symbols found in /usr/bin/gqrx)
(gdb) run
Starting program: /usr/bin/gqrx /usr/bin/gqrx
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff06a0640 (LWP 13019)]
[New Thread 0x7fffefe9f640 (LWP 13020)]
[New Thread 0x7fffef69e640 (LWP 13021)]
[New Thread 0x7fffeee9d640 (LWP 13022)]
[New Thread 0x7fffee69c640 (LWP 13023)]
[New Thread 0x7fffede9b640 (LWP 13024)]
[New Thread 0x7fffed69a640 (LWP 13025)]
[New Thread 0x7fffece99640 (LWP 13026)]
[New Thread 0x7fffec698640 (LWP 13027)]
[New Thread 0x7fffebe97640 (LWP 13028)]
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[New Thread 0x7fffead68640 (LWP 13029)]
[New Thread 0x7fffe92ba640 (LWP 13030)]
[New Thread 0x7fffe8ab9640 (LWP 13031)]
[New Thread 0x7fffdbfff640 (LWP 13032)]
[New Thread 0x7fffdb7fe640 (LWP 13033)]
[New Thread 0x7fffdade1640 (LWP 13034)]
[New Thread 0x7fffda5e0640 (LWP 13036)]
[Thread 0x7fffda5e0640 (LWP 13036) exited]
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
file_source :warning: file size is not a multiple of item size
Resampling audio 96000 -> 48000
[New Thread 0x7fffda5e0640 (LWP 13037)]
BandPlanFile is /home/eskindir/.config/gqrx/bandplan.csv
BookmarksFile is /home/eskindir/.config/gqrx/bookmarks.csv
[New Thread 0x7fffd881e640 (LWP 13038)]
[Thread 0x7fffd881e640 (LWP 13038) exited]
[New Thread 0x7fffd881e640 (LWP 13039)]
[Thread 0x7fffd881e640 (LWP 13039) exited]
[New Thread 0x7fffd881e640 (LWP 13040)]
[Thread 0x7fffd881e640 (LWP 13040) exited]
[New Thread 0x7fffd881e640 (LWP 13041)]
[New Thread 0x7fffc5ffd640 (LWP 13042)]
[New Thread 0x7fffc57fc640 (LWP 13043)]
[New Thread 0x7fffc4ffb640 (LWP 13044)]
[New Thread 0x7fffa7fff640 (LWP 13045)]
[New Thread 0x7fffa77fe640 (LWP 13046)]
[New Thread 0x7fffa6ffd640 (LWP 13047)]
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
[New Thread 0x7fffa67fc640 (LWP 13048)]
[New Thread 0x7fffa5ffb640 (LWP 13049)]
[New Thread 0x7fffa57fa640 (LWP 13050)]
[New Thread 0x7fffa4ff9640 (LWP 13051)]
[Thread 0x7fffa57fa640 (LWP 13050) exited]
[New Thread 0x7fff8ffff640 (LWP 13052)]
[New Thread 0x7fff8f7fe640 (LWP 13053)]
[New Thread 0x7fff8effd640 (LWP 13054)]
[Thread 0x7fffc4ffb640 (LWP 13044) exited]
[Thread 0x7fffc5ffd640 (LWP 13042) exited]
[Thread 0x7fffc57fc640 (LWP 13043) exited]
[Thread 0x7fffd881e640 (LWP 13041) exited]
[Thread 0x7fffa77fe640 (LWP 13046) exited]
[Thread 0x7fff8effd640 (LWP 13054) exited]
[Thread 0x7fffa7fff640 (LWP 13045) exited]
[Thread 0x7fffa67fc640 (LWP 13048) exited]
[Thread 0x7fff8ffff640 (LWP 13052) exited]
[... 136 lines cut ...]
[Thread 0x7fffa57fa640 (LWP 13090) exited]
[Thread 0x7fff70ff9640 (LWP 13088) exited]
[Thread 0x7fffc51ff640 (LWP 13065) exited]
[Thread 0x7fff737fe640 (LWP 13084) exited]
[Thread 0x7fff71ffb640 (LWP 13087) exited]
[Thread 0x7fffa5ffb640 (LWP 13049) exited]
[Thread 0x7fffa6ffd640 (LWP 13047) exited]
[Thread 0x7fffdade1640 (LWP 13034) exited]
[Thread 0x7fffede9b640 (LWP 13024) exited]
[Thread 0x7fffeee9d640 (LWP 13022) exited]
[Thread 0x7fffec698640 (LWP 13027) exited]
[Thread 0x7fffef69e640 (LWP 13021) exited]
[Thread 0x7fffed69a640 (LWP 13025) exited]
[Thread 0x7fffefe9f640 (LWP 13020) exited]
[Thread 0x7fffece99640 (LWP 13026) exited]
[Thread 0x7fffee69c640 (LWP 13023) exited]
[Thread 0x7fffebe97640 (LWP 13028) exited]
[Thread 0x7ffff06a0640 (LWP 13019) exited]
[Thread 0x7fffdbfff640 (LWP 13032) exited]
[Thread 0x7fffe8ab9640 (LWP 13031) exited]
[Thread 0x7fffe92ba640 (LWP 13030) exited]
[Thread 0x7ffff06a7e80 (LWP 13016) exited]
[Thread 0x7fffdb7fe640 (LWP 13033) exited]
[New process 13016]
[Inferior 1 (process 13016) exited normally]

bjvt, 2023-05-18:
I get the same problem, but running Gqrx 2.15.8 on ubuntu 22.04 LTS.  Cubic SDR does the same thing and the last lines are  below:
SDRThread::readStream(): 2. SoapySDR read failed with code: -1
SDRThread::readStream(): 2. SoapySDR read failed with code: -1
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
terminate called after throwing an instance of 'std::runtime_error'
  what():  setFrequency failed
Aborted (core dumped)

vladisslav2011, 2023-05-18:
@bjvt
LibUSB error -9 (EPIPE) usually means, that the device was disconnected during transfer.
It may be a bad cable, loose connection or insufficient power.
That's not a Gqrx issue.

nickoe, 2023-10-25:
Given there is no more updates on this, and it is a segault, an there is not backtrace attached. Maybe just close this bug by now?

argilo, 2023-10-31:
Even though the root cause appears to be outside Gqrx, I left this open because there appears to be a problem in Gqrx also:

```
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
```

Gqrx should not allow exceptions to bubble up to Qt.

vladtcvs, 2024-01-22:
Also have such issue when change frequency. I think in case of hardware problem Gqrx should show message, not crash

kengruven, 2025-02-06:
I'm experiencing this crash, too.

"It may be a bad cable, loose connection or insufficient power."

I'm connecting directly to a motherboard USB header, and the device only asks for 500 mA.  No other USB device has ever had trouble with any motherboard USB ports on my computer.  No other SDR application has an issue when using this SDR hardware.

I can run a waterfall display in gqrx indefinitely with no issues, but immediately experience this crash ("setFrequency failed") 100% of the time when I try to change the frequency.  Given the symptoms, it seems unlikely to me that this is an intermittent hardware issue.

I'm able to build gqrx and run it in gdb, but I'm not a C++ programmer so I'm not sure how to get a stack trace from inside Qt (the dialog box seems to catch it and block gdb).  If you tell me what to do, I'll be happy to do it.  I found a webpage that said to subclass QApplication and override `notify`, but the stack trace I get from breaking in that that only has Qt frames.