gqrx-sdr/gqrx #1138: Can't compile the GQRX on Ubuntu 20.04 -  gnuradio/buffer_reader.h: No such file or directory

atlex, 2022-08-08:
Hi.
I can't compile the Gqrx on Ubuntu 20.04.
My commands:

$ sudo add-apt-repository ppa:gnuradio/gnuradio-master
$ sudo apt update && sudo apt upgrade
$ sudo apt install gnuradio  gr-osmosdr qt5-default libqt5svg5-dev portaudio19-dev
$ git clone https://github.com/gqrx-sdr/gqrx.git 
$ cd gqrx
$ mkdir build
$ cd build
$ export CXXFLAGS=-O2
$ cmake ..
$ make

ERRORS:

```
Scanning dependencies of target gqrx
[ 24%] Building CXX object src/CMakeFiles/gqrx.dir/gqrx_autogen/mocs_compilation.cpp.o
In file included from /media/veracrypt1/gqrx/src/applications/gqrx/receiver.h:45,
                 from /media/veracrypt1/gqrx/src/applications/gqrx/mainwindow.h:48,
                 from /media/veracrypt1/gqrx/build/src/gqrx_autogen/JZP7SBZKMR/moc_mainwindow.cpp:9,
                 from /media/veracrypt1/gqrx/build/src/gqrx_autogen/mocs_compilation.cpp:2:
/media/veracrypt1/gqrx/src/dsp/rx_meter.h:29:10: fatal error: gnuradio/buffer_reader.h: No such file or directory
   29 | #include <gnuradio/buffer_reader.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/gqrx.dir/build.make:162: src/CMakeFiles/gqrx.dir/gqrx_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:248: src/CMakeFiles/gqrx.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

vladisslav2011, 2022-08-11:
Hello.

> No such file or directory

It looks, like GNU Radio is not on the include path.
Have you tried building Gqrx with GNU Radio version, shipped by your distribution instead of PPA version?

argilo, 2023-04-28:
I suspect the problem is that the `gnuradio-dev` package was not installed.

This does not appear to be a Gqrx problem, so I'll close off this issue.

closed 2023-04-28 by argilo