gqrx-sdr/gqrx #1143: Segmentation fault when working with bookmarks

yuvadm, 2022-08-16:
When working with bookmarks at some point this crash happens.

Versions: Arch Linux 5.18.16 + Qt 6.3.1

```
Thread 1 "gqrx" received signal SIGSEGV, Segmentation fault.
0x00007ffff67de8da in QString::operator=(QString const&) () from /usr/lib/libQt6Core.so.6
(gdb) bt
#0  0x00007ffff67de8da in QString::operator=(QString const&) () at /usr/lib/libQt6Core.so.6
#1  0x00007ffff67dec16 in QString::append(QString const&) () at /usr/lib/libQt6Core.so.6
#2  0x000055555565a4ee in BookmarksTableModel::data(QModelIndex const&, int) const ()
#3  0x00007ffff68eb421 in QAbstractItemModel::multiData(QModelIndex const&, QModelRoleDataSpan) const () at /usr/lib/libQt6Core.so.6
#4  0x00007ffff7832ff3 in QStyledItemDelegate::initStyleOption(QStyleOptionViewItem*, QModelIndex const&) const () at /usr/lib/libQt6Widgets.so.6
#5  0x00007ffff7836c30 in QStyledItemDelegate::paint(QPainter*, QStyleOptionViewItem const&, QModelIndex const&) const () at /usr/lib/libQt6Widgets.so.6
#6  0x00007ffff7859cb1 in  () at /usr/lib/libQt6Widgets.so.6
#7  0x00007ffff785f8af in QTableView::paintEvent(QPaintEvent*) () at /usr/lib/libQt6Widgets.so.6
#8  0x00007ffff75b82c0 in QWidget::event(QEvent*) () at /usr/lib/libQt6Widgets.so.6
#9  0x00007ffff7643b37 in QFrame::event(QEvent*) () at /usr/lib/libQt6Widgets.so.6
#10 0x00007ffff6711c88 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /usr/lib/libQt6Core.so.6
#11 0x00007ffff75749cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt6Widgets.so.6
#12 0x00007ffff6713088 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt6Core.so.6
#13 0x00007ffff75a9a9b in QWidgetPrivate::sendPaintEvent(QRegion const&) () at /usr/lib/libQt6Widgets.so.6
#14 0x00007ffff75ab557 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () at /usr/lib/libQt6Widgets.so.6
#15 0x00007ffff75c6d73 in  () at /usr/lib/libQt6Widgets.so.6
#16 0x00007ffff75b7fca in QWidget::event(QEvent*) () at /usr/lib/libQt6Widgets.so.6
#17 0x00007ffff75749dc in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt6Widgets.so.6
#18 0x00007ffff6713088 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt6Core.so.6
#19 0x00007ffff6714cd8 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib/libQt6Core.so.6
#20 0x00007ffff6932338 in  () at /usr/lib/libQt6Core.so.6
#21 0x00007ffff4104c6b in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#22 0x00007ffff415b001 in  () at /usr/lib/libglib-2.0.so.0
#23 0x00007ffff4102392 in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#24 0x00007ffff69304d2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt6Core.so.6
#25 0x00007ffff671c014 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt6Core.so.6
#26 0x00007ffff67166ab in QCoreApplication::exec() () at /usr/lib/libQt6Core.so.6
#27 0x00005555555eb157 in main ()
```

vladisslav2011, 2022-08-16:
Hello.
It's use after free...
Qt6 QList reallocates it's members internally using move constructor and invalidates pointers, stored in `BookmarkInfo::tags`.
I think, switching to using smart pointers instead of raw pointers for `Bookmarks::m_TagList` and `BookmarkInfo::tags` items would resolve this issue.

yuvadm, 2022-08-17:
@vladisslav2011 thanks for the... pointer :sunglasses: Is there any convention that gqrx prefers for smart pointers? If so, I'd gladly open a PR.

vladisslav2011, 2022-08-17:
If you are implementing a new GNU Radio block, you should check GNU Radio version and use either `boost::shared_ptr` or `std::shared_ptr` (not the case).
In any other case, you should use `std::shared_ptr` to not bring back hard dependency on boost.

yuvadm, 2022-08-17:
I'm not sure I understand the connection to GNU Radio. I want to help fix this issue in the gqrx codebase and I can't seem to find any usage of shared pointers outside of header files.

vladisslav2011, 2022-08-17:
It is better to ask @argilo about correct usage of shared pointers in this project.
I think, I've fixed this bug on top of my "dev" branch: https://github.com/vladisslav2011/gqrx/commit/16f778fea097733957b245dc2b55e99d1f177b3f
That's how it may look like...

vladisslav2011, 2022-08-24:
@yuvadm I've rebased my fix on top of master (#1145). Could you test and review it. Or open a PR with your fix if it looks better.

closed 2023-04-16 by argilo