gqrx-sdr/gqrx #1144: Segmentation fault with OpenSuse Tumbleweed hardware:sdr build

aaronautical, 2022-08-18:
I am getting a segmentation fault when launching gqrx version 2.15.9 installed from the OpenSuse Tumbleweed hardware:sdr repository. The gdb backtrace is attached. I previously posted about this issue in the google group. See [here](https://groups.google.com/g/gqrx/c/4IMdogpeVos) for a full history of what I have tried. In short, I have reinstalled Tumbleweed, but the issue persists. I did, however, successfully build and run version 2.15.9 from source. This leads me to believe there is an issue with the build in [hardware:sdr](https://download.opensuse.org/repositories/hardware:/sdr/openSUSE_Tumbleweed/).

[gdb.txt](https://github.com/gqrx-sdr/gqrx/files/9377870/gdb.txt)

argilo, 2022-08-19:
Unfortunately there's not much useful information in the stack trace. If you install the `-debuginfo` packages for Gqrx and its dependencies, then more information will likely become available.

aaronautical, 2022-08-19:
I have installed the `-debuginfo` packages for Gqrx and its dependencies; however, the gdb backtrace looks the same. It's possible I missed a package. Here is a list of the packages I installed:
```
gqrx-debuginfo
gnuradio-debuginfo
libgnuradio-3_10_3-debuginfo
libgnuradio-osmosdr0_2_0-debuginfo
gr-osmosdr-debuginfo
glibc-debuginfo
libstdc++6-debuginfo
libqt5core5-debuginfo
libqt5-qtbase-debugsource
libqt5gui5-debuginfo
libqt5widgets5-debuginfo
libqt5network5-debuginfo
libqt5svg5-debuginfo
libpulse0-debuginfo
libvolk2_5-debuginfo
volk-debuginfo
airspy-debuginfo
airspyhf-debuginfo
libairspy0-debuginfo
libairspyhf0-debuginfo
```

vladisslav2011, 2022-08-19:
[image]
I can't confirm. It just works.

argilo, 2022-08-20:
You could maybe try installing strace and running `strace gqrx` to see whether there are any clues about what's going on just before the crash.

The fact that installing `gqrx-debuginfo` didn't add anything to the stack trace would suggest that the problem is not in Gqrx itself, but perhaps in one of the SDR drivers.

aaronautical, 2022-08-20:
Here's the output of [strace gqrx.txt](https://github.com/gqrx-sdr/gqrx/files/9385989/strace.gqrx.txt).

I booted a live Tumbleweed image and installed gqrx from hardware:sdr, and I still got the segmentation fault. Since that worked for @vladisslav2011, maybe it's something specific to my hardware. I'm using the RTL-SDR blog SDR.

argilo, 2022-08-20:
Unfortunately I don't see much of interest in the strace log either. The crash appears to happen during loading of shared libraries, and before Gqrx or even GNU Radio begin initializing themselves.

Greenscreener, 2022-08-23:
Can confirm this bug on my install of OpenSUSE Tumbleweed too.

vladisslav2011, 2022-08-23:
> x86_64 20220820

I've tested i686 build. x86_64 build may contain different bugs. I'll retest on x86_64 livecd.

Greenscreener, 2022-08-23:
AppImage works normally.

argilo, 2023-04-29:
I suspect this was fixed in https://build.opensuse.org/request/show/1031623.

I installed gqrx from hardware:sdr on an x86_64 system without issue.

closed 2023-04-29 by argilo