gqrx-sdr/gqrx #1150: Feature request: bandplan overlay

hackery, 2022-10-07:
I constantly have to refer to source in other windows to identify what modes I should be expecting at each point along a band - would be very nice to have this embedded in the panadapter behind the frequency ticks:

[image]

Band plan data should be selectable from alternative files, and the colour mapping to modes selectable by the user.

argilo, 2022-10-08:
Gqrx already has a bandplan feature; it was added in https://github.com/gqrx-sdr/gqrx/pull/767. Have you tried that? Or are you proposing changes to it?

hackery, 2022-10-08:
!! I'd not seen that. Old release in my distro. I scanned the changelog but missed this. ty

closed 2022-10-08 by hackery