gqrx-sdr/gqrx #1152: When change frequency from Bookmarks it takes 8 seconds to switch

ja-Bertolin, 2022-10-09:
Hi

I am facing an issue when I choose a frequency from Bookmarks, when I click on one of the frequencies it takes 7 seconds to change the spectrum of the frequency and 1 second more to change the audio. Why it could be?

vladisslav2011, 2022-10-11:
Hello.
Do you have the same 7-second delay when tuning using main freqctrl over the panadapter?
Have you tried opening the "Configure I/O device" dialog and closing it with "OK" when the DSP is running (this should flush all buffers)?

ast, 2022-10-29:
@ja-Bertolin I'm currently working on a much improved Airspy HF+ driver for gr-osmosdr. I'm not sure if it will fix your particular issue but it will give you better performance, more features, and make it easier for me to track down your issue.

https://github.com/ast/gr-osmosdr