gqrx-sdr/gqrx #1154: Fail to start on Ubuntu 22.10: libgnuradio-funcube bad lib.

stdevPavelmc, 2022-10-28:
Tested on Gqrx from:

- Ubuntu repository
- App image (latest release from this repo)
- Src compiled on Ubuntu 22.10

It all fails to start with this output:

```
src/gqrx: error while loading shared libraries: libgnuradio-funcube.so.1.0.0: cannot open shared object file: No such file or directory
```

A search for the packager version on the system reveals that the version is:

` libgnuradio-funcube3.10.0`

And the lib file is named/placed on:

`/usr/lib/x86_64-linux-gnu/libgnuradio-funcube.so.3.10.0.0`

I can help with test, compile, etc, just ask

stdevPavelmc, 2022-10-28:
Compiling and installing funcube support from https://github.com/dl1ksv/gr-funcube make no difference.

I don't even have a funcube device, just a RTL, HackRF and a SDRPplay RSP1

stdevPavelmc, 2022-10-28:
Compiling `gr-osmosdr` from https://gitea.osmocom.org/sdr/gr-osmosdr **did the trick**.

So, for some reason the `gr-osmosdr-0.2.3-7` from the repository has a bad reference for the funcube lib, maybe?

Built app from source and tested ok on HackRF, RTL-SDR & SDRPlay RSP1; built appimage from source and tested ok on the just listed hardware.

Will let open for new Ubuntu 22.10 users to find. Owner feel free to close it when you like.

argilo, 2023-04-29:
Is this still an issue? I installed Ubuntu 22.10 in a virtual machine, installed Gqrx from the package manager, and it ran without issue.

argilo, 2023-05-07:
I haven't heard anything back, so I'll assume this is fixed. Please let me know if it's still an issue.

closed 2023-05-07 by argilo