gqrx-sdr/gqrx #1155: Fails to start on Ubuntu 22.10 arm64 Raspberry Pi 4

shull895, 2022-10-30:
After installation from snap store the program fails to start.
When installed from terminal it will stop because 'qt5-default' has no suitable replacement then stops.

argilo, 2023-04-30:
> After installation from snap store the program fails to start.

I would guess that this is a duplicate of #1154, which now appears to be fixed.

> When installed from terminal it will stop because 'qt5-default' has no suitable replacement then stops.

I'm guessing you were following the instructions at https://gqrx.dk/download/gqrx-sdr-for-the-raspberry-pi. They were out of date, so I've updated them to work with the latest version of Raspberry Pi OS. Hopefully they will work with Ubuntu as well.

argilo, 2023-04-30:
I'll close off this issue now, since I believe it is resolved. Let me know if not.

closed 2023-04-30 by argilo