gqrx-sdr/gqrx #1158: Build Problems

M0XFX, 2022-11-02:
Hi I can't seem to be able to build gqrx i get the following warnings and errors.

>>>>>>>>>>>>>>>>>>>>>
(following make command)
[100%] Linking CXX executable gqrx
/usr/bin/ld: warning: libgnuradio-blocks.so.3.10.1, needed by /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so, may conflict with libgnuradio-blocks.so.3.10.4
/usr/bin/ld: warning: libgnuradio-runtime.so.3.10.1, needed by /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so, may conflict with libgnuradio-runtime.so.3.10.4
/usr/bin/ld: warning: libgnuradio-pmt.so.3.10.1, needed by /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so, may conflict with libgnuradio-pmt.so.3.10.4
[100%] Built target gqrx
>>>>>>>>>>>>>>>>>>>>>

>>>>>>>>>>>>>>>>>>>>
(When I try to run gqrx) ./gqrx
terminate called after throwing an instance of 'std::runtime_error'
  what():  rpcmanager: Aggregator not in use, and a rpc booter is already registered
Aborted (core dumped)
>>>>>>>>>>>>>>>>>>>>

My conditions
ubuntu 22.04
gnu radio 3.10.4.0 and included dependencies (Python 3.10.6) installed which is working
cmake version 3.22.1
building from gqrx-2.15.9 download from github
Qt 5.15.3

I think there is a conflict between the gqrx and the latest version of gnuradio components. 

Any help would be much appreciated

vladisslav2011, 2022-11-02:
Hello.
You have conflicting versions of GNU Radio (3.10.4) and gr-osmosdr, that was built with different GNU Radio version (3.10.1).
Uninstall gr-osmosdr and build it from source or uninstall GNU Radio 3.10.4 and install GNU Radio 3.10.1.

M0XFX, 2022-11-03:
Thank you for your help and getting back to me so quickly, 

OK so have removed all elements of GNU radio and gr-osmosdr and I have now installed GNU radio 3.10.1.1 (Python 3.10.6) as described on the gnuradio.org web site (sudo apt-get install gnuradio).

Now when I go to build Cmake is says it can't find Gnuradio-osmosdr which I guess it to be expected. See following error>>>  

Could NOT find Gnuradio-osmosdr (missing: GNURADIO_OSMOSDR_LIBRARIES
GNURADIO_OSMOSDR_INCLUDE_DIRS)

My next thought is to install gr-osmosdr from source however; I'm concerned that it will cause the same issue again i.e.  looking for *3.10.4 library's. The reason I am hesitant to install  gr-osmosdr again as it was painful to remove last time.

Is there an older version of gr-osmosdr that the gqrx build is looking for or do I go ahead and download the latest version from "https://osmocom.org/projects/gr-gsm/wiki/Installation" and install from source again?

vladisslav2011, 2022-11-03:
```
git clone https://github.com/osmocom/gr-osmosdr
cd gr-osmosdr
mkdir build
cd build
cmake ..
make
```
If the build fails, disable failing components with `cmake .. -DENABLE_...=off` (or install missing dependencies) and rebuild
```
sudo make install
sudo ldconfig
```
Retry building Gqrx.

M0XFX, 2022-11-03:
I did as you suggested and cloned gr-osmosdr from github and I did have to install libsndfile-dev which i needed anyway to build it.

I now have no errors or warnings and a clean build of gqrx. 

Thank you so much for your help best regards.

[100%] Linking CXX executable gqrx
[100%] Built target gqrx

closed 2022-11-07 by argilo