gqrx-sdr/gqrx #1175: Inconsistent bookmark table behaviour for select/edit/activate (click, second-click and double-click)

pinkavaj, 2022-11-18:
I have found the following behavior when playing with Bookmarks tab and I perceive it as inconsistent

* double-click in `tags` column opens the tag list editor
* double-click in any other column activates the bookmark
* single-click on `tags` column does nothing
* single-click on already selected bookmark on columns Bandwidth, Modulation, Frequency and Name will start editing

IMHO any mouse action on any of the columns should do the same think (edit or activate the bookmark)

pinkavaj, 2022-11-18:
@argilo I might try to fix this, but it changes the user experience, is it somethink which have a real chance to get in? Asking to not waste time.

My opinion is to use the `single-click` on already selected row to start editing (default Qt behavior) and `double-click` to activate the bookmark.

vladisslav2011, 2022-11-19:
Editing a tag list after doubleclicking tags cell results in tuning to a bookmark. I don't think, that this should happen.

pinkavaj, 2022-11-19:
@vladisslav2011 So on your system the edit is triggered by double-click and not by single-click on already selected bookmark, right? So the behavior is may-be platform dependent? On what platform you are? I'm on Arch Linux i3 desktop environment. Latter I can check on KDE.

Editing is, except for the tags, handled by Qt internally, so only the bookmark activation is implemented in GQRX. What behaviour wold you expect? How should be bookmark activated?

vladisslav2011, 2022-11-19:
On my system editing is triggered by single-click on already selected bookmark, except tags cell.
The behavior is not platform dependent. It's Qt-dependent.
My platform is Ubuntu Mate 18.04 x86-64 + some testing chroots (Ubuntu 16.04 i586, Ubuntu 20.04 x86-64, Ubuntu 22.04 x86-64). No DE is installed in chroots.
I have never used tags, so I can't propose a better action to start editing tags, than current double click.
But I don't think, that editing tags list should activate a bookmark on confirmation.
I think, it may be better to leave current doubleclick as default activation action (tune and load settings?) and add 2 actions to a context menu with both options (tune only and tune and load settings).

pinkavaj, 2022-11-19:
I too think the double-click should be used for activation, what I consider an issue is the double-click on tags does not do activation but editing.

I also agree editing tags should not activate the bookmark on confirmation.