gqrx-sdr/gqrx #1177: [FEATURE REQUEST] Option for choosing input channel, for the device that has more than one

l1berat0r, 2022-12-02:
I'm using GQRX over soapy, with LimeSDR, which has 2 input channels and I'm only able to use one of them.
Maybe it would be a good idea, to add a channel selection option in the input configuration panel?

vladisslav2011, 2022-12-02:
Hello.
Have you tried setting `nchan` option?
https://community.libre.space/t/limesdr-switching-to-rx2-l-antenna-soapy-nchan-2-argument/6989

l1berat0r, 2022-12-04:
Hi,
Yes. I've made some tests, to verify what I've read on the internet and setting `nchan` doesn't select a channel, but it changes the number of output ports, so when I set `nchan=2` in gqrx, it crashes, because RX2 is not connected to anything.
I was able to compile a PoC in which I connected RX1 to a null sink and RX2 to the actual input in the signal route. I worked and allowed me to listen to the second channel. I'm not sure however is it the most optimal solution (will the data be feed through the unused channel as well?).

The other idea, that could be worth considering, would be adding one more configuration option to the `gr-osmocom` that would allow the source selection. I tried that as well and it seems very little trouble, but I don't know if the creators of the library will be willing to allow that.

Regardles of the option, it still would be nice a have a select box, that would allow a user, to switch between sources. What do you think?

willcode, 2024-03-22:
I just ran across this on a UHD with two front ends. Streaming both channels simultaneously would increase USB or network bandwidth, and gqrx would need to switch between channels on the fly when the antenna is changed.

A simpler solution for UHD at least is to specify the subdev, e.g., `subdev=B:0` to access the second front end.