gqrx-sdr/gqrx #1180: illegal instruction(core dump) ,in ubuntu 22.04

shuguanglangchao, 2022-12-29:
I update my ubuntu OS from 16.04 to 22.04,when I install Gqrx from Ubuntu Software,the gqrx does't work.
so I try to make from source code,the same .
why?

OOpeExtactOR, 2023-02-27:
I can confirm having the same error "Illegal instruction (core dumped)" on a fresh Lubuntu. 22.04.2 , kernel 5.15.0-60
console start shows this error, GUI start simply shows nothing

gcomeaux, 2023-03-06:
On my machine, it looks like the problem is that the SSE 4.2 instruction set extensions are missing, specifically, the pcmpgtq opcode. `cat /proc/cpuinfo` shows that the latest I have is sse4_1. If pcmpgtq is really required, I guess the only option is to update my hardware. Otherwise, the compiler should be checked for generating less compatible instructions.

argilo, 2023-03-06:
Could someone use gdb to provide a full stack trace, preferably with debug symbols installed? This would help track down whether this is an issue in Gqrx itself, or one of its dependencies.

gcomeaux, 2023-03-07:
Looks like the problem is with librte-net22 package?
`
$ dpkg -S /usr/lib/x86_64-linux-gnu/librte_net.so.22.0
librte-net22:amd64: /usr/lib/x86_64-linux-gnu/librte_net.so.22.0
`

`
Downloading 0.01 MB separate debug info for /lib/x86_64-linux-gnu/libbrotlicommon.so.1

Program received signal SIGILL, Illegal instruction.
0x00007ffff341b684 in __rte_rand_lfsr258_gen_seed (min_value=<optimized out>, seed=<optimized out>) at ../lib/eal/common/rte_random.c:60
Download failed: Invalid argument.  Continuing without source file ./obj-x86_64-linux-gnu/../lib/eal/common/rte_random.c.
60      ../lib/eal/common/rte_random.c: No such file or directory.
(gdb) bt
#0  0x00007ffff341b684 in __rte_rand_lfsr258_gen_seed (min_value=<optimized out>, seed=<optimized out>)
    at ../lib/eal/common/rte_random.c:60
#1  __rte_srand_lfsr258 (state=<optimized out>, seed=<optimized out>) at ../lib/eal/common/rte_random.c:72
#2  rte_srand (seed=7808623144792863306) at ../lib/eal/common/rte_random.c:86
#3  0x00007ffff340f8cc in _sub_I_65535_0.0 () from /lib/x86_64-linux-gnu/librte_eal.so.22
#4  0x00007ffff7fcbfbe in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffe2c8,
    env=env@entry=0x7fffffffe2d8) at ./elf/dl-init.c:70
#5  0x00007ffff7fcc0a8 in call_init (env=0x7fffffffe2d8, argv=0x7fffffffe2c8, argc=1, l=<optimized out>)
    at ./elf/dl-init.c:33
#6  _dl_init (main_map=0x7ffff7ffe2e0, argc=1, argv=0x7fffffffe2c8, env=0x7fffffffe2d8) at ./elf/dl-init.c:117
#7  0x00007ffff7fe48b0 in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#8  0x0000000000000001 in ?? ()
#9  0x00007fffffffe55c in ?? ()
#10 0x0000000000000000 in ?? ()
(gdb) df __rte_rand_lfsr258_gen_seed
Undefined command: "df".  Try "help".
(gdb) disassemble __rte_rand_lfsr258_gen_seed
Dump of assembler code for function rte_srand:
   0x00007ffff341b560 <+0>:     endbr64
   0x00007ffff341b564 <+4>:     movq   %rdi,%xmm2
   0x00007ffff341b569 <+9>:     movdqa 0x1b4bf(%rip),%xmm3        # 0x7ffff3436a30
   0x00007ffff341b571 <+17>:    punpcklqdq %xmm2,%xmm2
   0x00007ffff341b575 <+21>:    lea    0x48684(%rip),%rax        # 0x7ffff3463c00 <rand_states>
   0x00007ffff341b57c <+28>:    movaps %xmm2,-0x18(%rsp)
   0x00007ffff341b581 <+33>:    movdqa 0x1b486(%rip),%xmm13        # 0x7ffff3436a10
   0x00007ffff341b58a <+42>:    movdqa 0x1b4ae(%rip),%xmm2        # 0x7ffff3436a40
   0x00007ffff341b592 <+50>:    lea    0x2000(%rax),%rdx
   0x00007ffff341b599 <+57>:    movdqa 0x1b4be(%rip),%xmm14        # 0x7ffff3436a60
   0x00007ffff341b5a2 <+66>:    nopw   0x0(%rax,%rax,1)
   0x00007ffff341b5a8 <+72>:    movdqa %xmm13,%xmm1
   0x00007ffff341b5ad <+77>:    movdqa -0x18(%rsp),%xmm7
   0x00007ffff341b5b3 <+83>:    paddd  0x1b464(%rip),%xmm13        # 0x7ffff3436a20
   0x00007ffff341b5bc <+92>:    add    $0x100,%rax
   0x00007ffff341b5c2 <+98>:    movdqa 0x1b485(%rip),%xmm11        # 0x7ffff3436a50
   0x00007ffff341b5cb <+107>:   pmovzxdq %xmm1,%xmm4
   0x00007ffff341b5d0 <+112>:   psrldq $0x8,%xmm1
   0x00007ffff341b5d5 <+117>:   pmovzxdq %xmm1,%xmm1
   0x00007ffff341b5da <+122>:   paddq  %xmm7,%xmm4
   0x00007ffff341b5de <+126>:   movdqa 0x1b489(%rip),%xmm10        # 0x7ffff3436a70
   0x00007ffff341b5e7 <+135>:   paddq  %xmm7,%xmm1
   0x00007ffff341b5eb <+139>:   movdqa %xmm4,%xmm0
   0x00007ffff341b5ef <+143>:   movdqa 0x1b478(%rip),%xmm9        # 0x7ffff3436a70
   0x00007ffff341b5f8 <+152>:   movdqa %xmm1,%xmm5
   0x00007ffff341b5fc <+156>:   psrlq  $0x20,%xmm0
   0x00007ffff341b601 <+161>:   shufps $0x88,%xmm1,%xmm4
   0x00007ffff341b605 <+165>:   movdqa 0x1b472(%rip),%xmm8        # 0x7ffff3436a80
--Type <RET> for more, q to quit, c to continue without paging--
   0x00007ffff341b60e <+174>:   psrlq  $0x20,%xmm5
   0x00007ffff341b613 <+179>:   movdqa 0x1b465(%rip),%xmm7        # 0x7ffff3436a80
   0x00007ffff341b61b <+187>:   shufps $0x88,%xmm5,%xmm0
   0x00007ffff341b61f <+191>:   pxor   %xmm4,%xmm0
   0x00007ffff341b623 <+195>:   pmulld %xmm3,%xmm0
   0x00007ffff341b628 <+200>:   paddd  %xmm2,%xmm0
   0x00007ffff341b62c <+204>:   movdqa %xmm0,%xmm1
   0x00007ffff341b630 <+208>:   pmovzxdq %xmm0,%xmm4
   0x00007ffff341b635 <+213>:   pmulld %xmm3,%xmm1
   0x00007ffff341b63a <+218>:   psrldq $0x8,%xmm0
   0x00007ffff341b63f <+223>:   pmovzxdq %xmm0,%xmm0
   0x00007ffff341b644 <+228>:   paddd  %xmm2,%xmm1
   0x00007ffff341b648 <+232>:   pmovzxdq %xmm1,%xmm5
   0x00007ffff341b64d <+237>:   psllq  $0x20,%xmm5
   0x00007ffff341b652 <+242>:   por    %xmm4,%xmm5
   0x00007ffff341b656 <+246>:   movdqa %xmm1,%xmm4
   0x00007ffff341b65a <+250>:   paddq  %xmm5,%xmm11
   0x00007ffff341b65f <+255>:   pmulld %xmm3,%xmm1
   0x00007ffff341b664 <+260>:   psrldq $0x8,%xmm4
   0x00007ffff341b669 <+265>:   pmovzxdq %xmm4,%xmm4
   0x00007ffff341b66e <+270>:   psllq  $0x20,%xmm4
   0x00007ffff341b673 <+275>:   paddd  %xmm2,%xmm1
   0x00007ffff341b677 <+279>:   por    %xmm0,%xmm4
   0x00007ffff341b67b <+283>:   movdqa %xmm5,%xmm0
   0x00007ffff341b67f <+287>:   psubq  %xmm14,%xmm0
=> 0x00007ffff341b684 <+292>:   pcmpgtq 0x1b423(%rip),%xmm0        # 0x7ffff3436ab0
   0x00007ffff341b68d <+301>:   pblendvb %xmm0,%xmm5,%xmm11
   0x00007ffff341b693 <+307>:   movdqa %xmm4,%xmm0
   0x00007ffff341b697 <+311>:   movdqa 0x1b3b1(%rip),%xmm5        # 0x7ffff3436a50
`

argilo, 2023-03-07:
Yes, it does appear the problem lies in the librte-net22 package.

On Ubuntu 22.04, gqrx-sdr depends on gr-osmosdr, which depends on libuhd4.1.0 (to provide support for USRP radios), which depends on librte-hash22, which depends on librte-net22. So the problem would seem to be many layers removed from Gqrx itself. I would suggest opening a bug report on Ubuntu's librte-net22 package.

argilo, 2023-03-07:
You could perhaps verify that the problem is outside Gqrx by installing the `uhd-host` package and running `uhd_usrp_probe`, which also uses librte-net22.

gcomeaux, 2023-03-07:
I installed uhd-host, ran `gdb uhd_usrp_probe`, and I see the same problem. Thank you for the test case. I'll reference this issue over at the librte-net22 package.

argilo, 2023-03-07:
Thanks for checking. That confirms that the problem is outside Gqrx, but it's certainly unfortunate for us.

OOpeExtactOR, 2023-03-12:
Thanks for checking! It seems to be an issue of outdated HW, by all I can get.
Therefore, I understand if a fix will not be provided.

argilo, 2023-03-12:
I dug a bit more into this, and it seems like Debian provides two versions of libuhd, one which requires DPDK (https://packages.debian.org/sid/libuhd4.3.0-dpdk) and one which does not (https://packages.debian.org/sid/libuhd4.3.0). Only the DPDK version links to librte-*.

Likewise, Ubuntu 22.04 has two versions: https://packages.ubuntu.com/jammy/libuhd4.1.0-dpdk and https://packages.ubuntu.com/jammy/libuhd4.1.0. But they both link to DPDK libraries (librte-*). So it seems to me the bug lies with the libuhd4.1.0 package, which shouldn't use DPDK.

vsv68, 2023-06-19:
Solved the problem, assembled and installed uhd 4.3.0 from the source.

$ git clone https://git.launchpad.net/ubuntu/+source/uhd
$ cd uhd
$ cd host
$ mkdir build
$ cd build
$ cmake ..
$ make
$ sudo make install
$ cd /usr/lib/x86_64-linux-gnu/ 
$ sudo mv libuhd.so.4.1.0 /home/user/libuhd.so.4.1.0.back
$ sudo ln -s /usr/local/lib/libuhd.so.4.3.0 libuhd.so.4.1.0
$ sudo ldconfig

After that everything works.

AsciiWolf, 2023-07-24:
Please, report the issue [here](https://launchpad.net/ubuntu/+source/uhd/+bugs) instead of just working it around. Thanks!

jackcarrozzo, 2025-03-02:
For everyone else who finds this thread trying to solve the issue for UHD, with or without gqrx-

Hit this on freshly installed debian 12.9 vm running on an older Operton machine. Most of the items in uhd's `make test` failed due to Illegal Instruction with traps logged to `dmesg`.

I can confirm that building UHD from source and disabling DPDK does indeed work- reportedly this removed the offending code in `librte_net.so`.

```
git clone https://github.com/EttusResearch/uhd.git
cd uhd
git checkout UHD-4.8
cd host
mkdir build
cd build/
cmake -DENABLE_DPDK=OFF .. 
make -j12
make test
sudo make install
sudo ldconfig
cd python
sudo ./setup.py build
sudo ./setup.py install
```
and lo,
```
$ uhd_config_info --print-all
UHD 4.8.0.0-0-g308126a4
Build date: Sun, 02 Mar 2025 04:12:25
C compiler: GNU 12.2.0
C++ compiler: GNU 12.2.0
C flags: -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_ASIO_HAS_STD_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DHAVE_CONFIG_H -DUHD_LOG_MIN_LEVEL=1 -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_CONSOLE_COLOR
C++ flags: -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_ASIO_HAS_STD_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DHAVE_CONFIG_H -DUHD_LOG_MIN_LEVEL=1 -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_CONSOLE_COLOR -fvisibility=hidden -fvisibility-inlines-hidden
Enabled components: LibUHD, LibUHD - C API, LibUHD - Python API, Examples, Utils, Tests, Python Module (Utils only), USB, USRP1, USRP2, MPMD, SIM, OctoClock
Install prefix: /usr/local
Boost version: 1.74
DPDK version: N/A
Libusb version: 1.0.27
Library path: /usr/local/lib
Package path: /usr/local
Package data path: /usr/local/share/uhd
Images directory: 
ABI version string: 4.8.0
```