gqrx-sdr/gqrx #1181: [Bug] gqrx did not start or crash, just hang

unknown10777, 2023-01-11:
$ time gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx 
Resampling audio 96000 -> 48000
BandPlanFile is /home/ayaka/.config/gqrx/bandplan.csv
BookmarksFile is /home/ayaka/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 12.2.0; Boost_107400; UHD_4.3.0.0+ds1-3
^C

real    14.57s
user    0.14s
sys     0.08s
cpu     1%

vladisslav2011, 2023-01-11:
Hello.
Could you start Gqrx under gdb:
```
gdb gqrx
run
```
Wait 10...15 seconds.
Break it with Ctrl+C.
Obtain a backtrace:
```
bt
```
Post the backtrace here.

P.S.
Running Gqrx with a clean config `gqrx -r` may help.

unknown10777, 2023-01-11:
```
$ uname -a               
Linux Akasha 5.18.0-kali7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.16-1kali1 (2022-08-31) x86_64 GNU/Linux
```

vladisslav2011, 2023-01-11:
It hangs trying to enumerate URP devices... Maybe one of your USB devices is buggy.
Does `lsusb` freeze too?

vladisslav2011, 2023-01-11:
Could you
1. Disconnect all USB devices (excluding mouse/keyboard) and disconnect from the network
2. Reboot the machine
3. Start Gqrx again
To make sure, that this bug is not related to a buggy USB device.

unknown10777, 2023-01-11:
Thank you for patient helping me.

closed 2023-01-11 by unknown10777