gqrx-sdr/gqrx #1183: Add AM-Sync demod options

sibrat, 2023-01-17:
It would be nice for better listening experience (preventing whistle) add pll bandwidth option and soft mute feature (reducing audio output level if pll not locked)

argilo, 2023-04-28:
> add pll bandwidth option

This option already exists in the "Mode options" panel for AM-Sync:

[image]

sibrat, 2023-04-29:
Probably I didn't formulate it correctly. What is implemented is the width of the feedback loop i.e. reaction speed. What i mean is to tune the width of the capture bandwidth (currently it's not limited and equals receiving (demodulating) bandwidth). It needs to be limited to prevent attempts of capturing adjacent channel carrier in case when signal of interest is weak and fading. Also there is cases when transmitter failed for some reason and then restarted. narrow capture bandwidth will ensure quick resync. (current "fast pll" option also will do fast resync, but it also quickly lose syncronisation so sync demodulator loses its pupose - to regenerate stable carrier in selective fading conditions.) 
best options for SWLing would be: current "medium pll" feedback speed and limiting capturing bandwidth to plus/minus 200Hz from tuning frequency.

argilo, 2023-04-29:
Ah, thanks for providing more detail.

Currently the PLL is limited to +/- 5 kHz, which seems excessive:

https://github.com/gqrx-sdr/gqrx/blob/386c18df667d195498910bdc73a574080dd4acab/src/dsp/rx_demod_am.cpp#L114

https://github.com/gqrx-sdr/gqrx/blob/386c18df667d195498910bdc73a574080dd4acab/src/dsp/rx_demod_am.cpp#L123-L125

I guess this should be lowered. I'm not sure whether it's necessary to make it configurable.

sibrat, 2023-05-02:
idk if there is other use cases, some LEO satellite reception maybe? If there is none then indeed this value can be just lowered.