gqrx-sdr/gqrx #1187: error due to alias functions "disconnect" with different number  of parameters (and probably different meaning"

bonne-soeur, 2023-02-07:
When installing on ubuntu 20.04lts from version argilo november 2022:
"cmake .." step is okey
but 
"make" produces when building object "receiver.cpp" 
a lot of errors the first of which is:
[ 28%] Building CXX object src/CMakeFiles/gqrx.dir/applications/gqrx/receiver.cpp.o
/home/yves/Installations/gqrx_vladisslav/src/applications/gqrx/receiver.cpp: In member function ‘void receiver::set_input_device(std::string)’:
/home/yves/Installations/gqrx_vladisslav/src/applications/gqrx/receiver.cpp:203:46: error: no matching function for call to ‘gr::top_block::disconnect(osmosdr::source::sptr&, int, fir_decim_cc_sptr&, int)’
  203 |         tb->disconnect(src, 0, input_decim, 0);
      |                                              ^
In file included from /usr/include/gnuradio/top_block.h:15,
                 from /home/yves/Installations/gqrx_vladisslav/src/applications/gqrx/receiver.cpp:30:
/usr/include/gnuradio/hier_block2.h:127:10: note: candidate: ‘void gr::hier_block2::disconnect(gr::basic_block_sptr)’
  127 |     void disconnect(basic_block_sptr block);
      |          ^~~~~~~~~~

argilo, 2023-02-08:
The compilation error appears similar to the ones reported in #941, #952, and #971. But unfortunately it seems nobody got to the bottom of it. Compilation also works fine on Ubuntu 20.04 in Gqrx's CI.

Did you install the `gnuradio-dev` and `gr-osmosdr` dependencies from apt? Did you build either GNU Radio or gr-osmosdr from source?

argilo, 2023-02-08:
A full build log would also be helpful.

bonne-soeur, 2023-02-08:
Dear Argilo, 
Thank you very much for your care.
Here is where I reach after some work.
The overloading of function connect() or disconnect() is done on purpose.
We have two candidates, one has 2 parameters, the other one has 4.
The one form that is called by the function receiver.cpp is the 4 parameters one.
It should fit, but it did not for the reason, I explained in the next minute.
The other candidate works with 2 parameters, and does not fit neither: that's normal.
So what is the problem with the 4 parameters candidate.
The calling function from gqrx/receiver.cpp and the prototype from gnuradio/hier_block2.h
use as parameters #1 and #3 shared pointer. But they are not of the same type!
receiver.cpp through osmosdr/source.h uses boost::shared_ptr type
whereas gnuradio/hier_block2.h uses std::shared_ptr type instead which are declered in gnuradio/runtime_types.h.
So I aligned source.h tptr to std::shared_ptr and it works.
Now, I am stuck with a final problem at the very end of "make":
make[2]: ***  Aucune règle pour fabriquer la cible « /usr/lib/x86_64-linux-gnu/liborc-0.4.so », nécessaire pour « src/gqrx ». Arrêt.
make[1]: *** [CMakeFiles/Makefile2:248 : src/CMakeFiles/gqrx.dir/all] Erreur 2
make: *** [Makefile:130 : all] Erreur 2
no rule to build the target  "/usr/lib/x86_64-linux-gnu/liborc-0.4.so "
Let us see what can be done
I hope to hear from your remarks
Yves

argilo, 2023-02-08:
> receiver.cpp through osmosdr/source.h uses boost::shared_ptr type whereas gnuradio/hier_block2.h uses std::shared_ptr type instead

It sounds to me like you are using incompatible versions of GNU Radio and gr-osmosdr. GNU Radio 3.8 uses boost::shared_ptr for block pointers, while 3.9+ switched to std::shared_ptr. The gr-osmosdr project made this chang as well, but perhaps you are using a version of gr-osmosdr designed for use with GNU Radio 3.8.

Ubuntu 20.04 comes with GNU Radio 3.8. Did you install a more recent version from source or from a PPA?

bonne-soeur, 2023-02-08:
Yes , You are right. I was intending to use the SDR-RSP1 clone with gqrx and so I asked to Vladisla2011 how he could be doing so , and then to use his new version of libmiris, and then I found that my gnuradio version was getting too old, and I was pushed to use a newer gnuradio and on monday (two days ago) I have installed gnuradio 3.10.5
So we see tht gnuradio is back to std::shared_ptr

argilo, 2023-02-08:
In that case, you will have to remove the `gr-osmosdr` package from your system and build gr-osmosdr from source.

bonne-soeur, 2023-02-08:
I undersatnd very well
but then which one is the fresh osmosdr. I also reinstalled osmosdr on monday, but probably not with the right one!

bonne-soeur, 2023-02-08:
I saw you also prepared an appimage . Thta's a great job!

argilo, 2023-02-08:
That's right. It includes all dependencies, so you don't need to worry about having a working GNU Radio & gr-osmosdr installation.

Or if you don't care about having the very latest version of Gqrx, you can install from the package manager with `sudo apt install gqrx-sdr`.

I'll close off this issue, since the problem was not in Gqrx itself.

closed 2023-02-08 by argilo

bonne-soeur, 2023-02-08:
the only left problem with gqrx  is about liborc-0.4.so that I dont finf!

bonne-soeur, 2023-02-08:
the gqrx from the ubuntu package manager was unable to receive the SDR-RSP1 and so I was invited to recompile everything with the library libmiris4

argilo, 2023-02-08:
You can install that dependency with `sudo apt install liborc-0.4-dev`.

bonne-soeur, 2023-02-08:
Thank you a lot for your help , at first  I miswrote liborc-0.4-dev 
Now it "make" to the end I just have a warning at the end , I think it is about the point you made about using "Volk"