gqrx-sdr/gqrx #1191: AM Mode no sound - All other modes including LSB USB OK despite volk_profile being run - SDRPlay RSP1 on RPi 4 B

vjkanth, 2023-02-20:
Have run volk_profile based on a suggestion but that has not fixed the problem. All other modes work as they should.
RPi Image was taken directly from SDRPlay.com site which had GQRX, gnuradio, cubicsdr etc pre-installed. I also installed SDR++ on the same build and can run all modes flawlessly. Have tried Upgrade to the latest of all libs but hasn't fixed this issue. Anyone face this? Any suggestions?

vladisslav2011, 2023-02-22:
Hello.
> Any suggestions?

1. Check gain slider (audio dock) - set the gain to -6db...0db
2. Check AGC mode/settings (receiver options dock) - the AGC should be enabled and configured correctly

vjkanth, 2023-02-23:
> Hello.
> 
> > Any suggestions?
> 
> 1. Check gain slider (audio dock) - set the gain to -6db...0db
> 2. Check AGC mode/settings (receiver options dock) - the AGC should be enabled and configured correctly

Tried both no joy 😠

vjkanth, 2023-02-23:
> > Hello.
> > > Any suggestions?
> > 
> > 
> > 
> > 1. Check gain slider (audio dock) - set the gain to -6db...0db
> > 2. Check AGC mode/settings (receiver options dock) - the AGC should be enabled and configured correctly
> 
> Tried both no joy 😠

Found the problem. Had to uncheck the DC removal option (next to the Mode selection) to hear audio. doh!

argilo, 2023-04-28:
I'm glad to hear you found a solution.

I had a look at the SDRplay Raspberry Pi image, and saw that it is using a very old pre-release version of Gqrx from May 2019. A lot has changed since then, so it is possible the issue you ran into is no longer present in the current Gqrx version (2.16). I haven't seen the problem myself, so I'll close off this issue for now. Feel free to re-open if it turns out that the bug is still present.

closed 2023-04-28 by argilo

argilo, 2023-04-29:
I was able to reproduce the problem, and it's almost certainly a duplicate of #570, so I'll investigate and post details there.

argilo, 2023-04-29:
Confirmed that this is a duplicate of #570; I've explained the root cause in https://github.com/gnuradio/volk/issues/622.

As a workaround, you could put the following in `~/.volk/volk_config`:

```
volk_32fc_magnitude_32f generic generic
```

This will prevent the broken VOLK kernel from being used.