gqrx-sdr/gqrx #1192: provide a means to change receiver frequency by mousewheel and (or) hotkeys without moving (and thus tearing) waterfall.

sibrat, 2023-02-27:
There is no way to change receiver frequency (not hardware frequency) rather than pinpointing it with left mouse key. It would be much more convinient do that with combination like shift + mouse wheel and page-up/page down key or maybe even make it default action and change hardware frequency with shift+scroll. Implementing issue #71 in this context also would be very nice as we are already have channel step in bandplan.

argilo, 2023-03-06:
The receiver frequency can be adjusted with the mouse wheel. Make sure the mouse pointer is hovering over the panadapter, waterfall, or digits in the "Receiver Options" tab.

sibrat, 2023-03-07:
> The receiver frequency can be adjusted with the mouse wheel. Make sure the mouse pointer is hovering over the panadapter, waterfall, or digits in the "Receiver Options" tab.

this indeed changes receiver IC frequency (in my case MSI001 tuning frequency), but i'm talking about receiving (demodulation) frequency

argilo, 2023-03-07:
I'm not sure why that would be. For me it changes the demodulation frequency. I tested with an RTL-SDR.

sibrat, 2023-03-07:
oh, i see. mouse wheel over tuning frequency changes hardware frequency, over waterfall changes demodulation frequency. Can't believe i never try this. my fault :)

closed 2023-03-07 by sibrat

argilo, 2023-03-07:
That explains it. I'm glad to hear it's working for you now.

sibrat, 2023-03-07:
Yes, it's very easy to overlook what exactly changes when baseband bandwidth is much wider than couple of megaherz, so maybe it worth it to somehow make presence of this difference more obvious. just suggestion...