gqrx-sdr/gqrx #1194: OrangePi after build not open

drozdi70, 2023-03-12:
Hi
I have build gqrx according to instrucyions on my OrangePI5 (Orange Pi 1.1.4 Bullseye with Linux 5.10.110-rockchip-rk3588)
And once runnig gqrx I see:
orangepi@orangepi5:~/work$ gqrx
qt.qpa.xcb: XKeyboard extension not present on the X server
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
BandPlanFile is /home/orangepi/.config/gqrx/bandplan.csv
BookmarksFile is /home/orangepi/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
QFSFileEngine::open: No file name specified
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '5'.
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '5'.
The X11 connection broke: Unsupported extension used (code 2)
X connection to :1.0 broken (explicit kill or server shutdown).
Kindly

vladisslav2011, 2023-03-12:
Hello.

>qt.qpa.xcb: XKeyboard extension not present on the X server

Are you using x11vnc headless setup?
If so, try running it on a local X11 server or use ssh X11 forwarding (slow and requires some pulseaudio tweaks).

drozdi70, 2023-03-12:
HiPlease close that issue.I took an image from the official page but it seems it was a bad image (mine was from 8.03.20023) :(It has been changed 2 days ago. I downloaded new one and the program was compiled and it seems to be opening now.KindlyRobertTemat: Re: [gqrx-sdr/gqrx] On orangePi after build not open (Issue #1194)Data: 2023-03-12 16:25Nadawca: "vladisslav2011" ***@***.***&gt;Adresat: "gqrx-sdr/gqrx" ***@***.***&gt;; DW: "drozdi70" ***@***.***&gt;; "Author" ***@***.***&gt;; 

Hello.

qt.qpa.xcb: XKeyboard extension not present on the X server

Are you using x11vnc headless setup?
If so, try running it on a local X11 server or use ssh X11 forwarding (slow and requires some pulseaudio tweaks).

drozdi70, 2023-03-12:
Hi
Please close that issue.
I took an image from the official page but it seems it was a bad image (mine was from 8.03.20023) :(
It has been changed 2 days ago. I downloaded new one and the program was compiled and it seems to be opening now.
Kindly

closed 2023-03-12 by drozdi70