gqrx-sdr/gqrx #1205: iq_swap_cc expensive when off

willcode, 2023-03-23:
The `iq_swap` block is always inline. When swap is turned off, samples are sent through a c2f then f2c block, to no effect. These blocks, together, also count for a large portion of the processor utilizaiton, whether or not they are acting as a pass-through.

https://github.com/gqrx-sdr/gqrx/blob/5588106e8ee2c8c20ada759a2199876e8f54803d/src/dsp/correct_iq_cc.cpp#L112

Make it possible to bypass the `iq_swap` block. The `dc_corr` already requires reconfiguration when the DC Cancel box is checked, so why not save a lot of processor in the most common "not swaped" case.

vladisslav2011, 2023-03-23:
Here is a PR https://github.com/gqrx-sdr/gqrx/pull/1063 that may solve this issue.

willcode, 2023-03-23:
#1206 is simpler. BTW, compilers are really good these days, so using GR blocks to implement simple math doesn't work out, even with volk.

vladisslav2011, 2023-03-23:
#1206 should be faster when IQ swap is enabled, but #1063 is faster when IQ swap is disabled. #1063 avoids buffer allocation and copy when IQ swap is disabled. At high sample rates even memcpy may consume a lot of CPU time.

willcode, 2023-03-23:
A combination could be the right thing, then. I don't think we're stressing out the memory bus on modern computer here, but I agree that fewer copies is better.

There was a note in the code about iq_correct not having an on/off switch, so I figured the goal might be to have fewer reconfigurations. Doesn't matter either way to me.

I also noticed that gr-osmosdr inserts `iqbalance:fix_cc` in its path unconditionally if gr-iqbalance is found. So I rebuilt gr-osmosdr with that cmake variable turned off ... one less block inline.

vladisslav2011, 2023-03-24:
The memory bandwidth is not a problem compared to the load from context switches when Gqrx is fed with small blocks at high rate. It becomes really problematic on a weak system like Raspberry Pi or my test laptop (Samsung NC10). It is better to have as less blocks running at full rate, as possible.

> I also noticed that gr-osmosdr inserts `iqbalance:fix_cc`

Me too.
Sometimes IQ correction is useful to me, so I have fixed this problem here: https://github.com/vladisslav2011/gr-osmosdr/tree/iq_balance_master
Feel free to cherry-pick...

willcode, 2023-03-24:
My solution was a little more brutal. Generally, I like to run stock anyway (except for Gqrx), to turn up any problems due to GR changes.

PR #1206 (2023-04-13): iq_swap_cc: implement directly instead of as a hier block
Much faster, especially when off, and fewer buffers.

Saves about a half a core on my system at 56 MB/s.

Fixes #1205
closed 2023-04-13 by argilo