gqrx-sdr/gqrx #1215: make error !

hm123tyt, 2023-03-30:
when i have ‘make’,
make[2]: *** [src/CMakeFiles/gqrx.dir/build.make:1201：src/gqrx] error 1
make[1]: *** [CMakeFiles/Makefile2:233：src/CMakeFiles/gqrx.dir/all] error 2

argilo, 2023-03-30:
Unfortunately, this is not enough information to diagnose the problem. Please provide the list of commands you executed when building, as well as the full output of each command.

hm123tyt, 2023-03-30:
[image]

argilo, 2023-03-30:
It appears that you have two conflicting versions of GNU Radio installed (3.9.8 & 3.10.1). You should uninstall one of those before attempting to build Gqrx and its dependencies.

Or instead of building from source, you could install Gqrx from the package manager with `sudo apt install gqrx-sdr`.

argilo, 2023-03-30:
I'll close this issue off for now, since it doesn't appear to be a problem with Gqrx.

closed 2023-03-30 by argilo