gqrx-sdr/gqrx #1216: make error again！

hm123tyt, 2023-03-31:
"I reinstalled ubuntu and followed the steps on github to install gqrx, but the following error occurred while making. Can you tell me how to solve the problem?"
[image]

argilo, 2023-03-31:
Did you build GNU Radio from source? If so, it appears that the WAV File Source & Sink blocks are not present. Those blocks are only built if the `libsndfile-dev` package is installed. You should install that package, then rebuild GNU Radio.

This does not appear to be a problem with Gqrx, so I'll close off this issue.

In the future, I would suggest asking questions like this in the support forum: https://groups.google.com/g/gqrx

closed 2023-03-31 by argilo

hm123tyt, 2023-03-31:
That's OK. Thank you very much!