gqrx-sdr/gqrx #1217: Start playing radio on startup?

doronbehar, 2023-04-05:
Hello!

Is it possible to start playing radio straight on startup? Or alternatively, control it remotely via UDP? I'm using Gqrx on an embedded system and I want it to start playing radio on startup, or at least via a remote command. Is this possible? Reading [remote-control.txt](https://github.com/gqrx-sdr/gqrx/blob/master/resources/remote-control.txt) doesn't list such command if I understand correctly.

doronbehar, 2023-04-07:
Oh I got it now! It's `u DSP` and `U DSP 1/0` commands! I misunderstood the "DSP" terminology when I initially read `remote-control.txt`, but looking at the source code helped me. Perhaps that could be improved a little?

```diff
diff --git i/resources/remote-control.txt w/resources/remote-control.txt
index 0964ec4..c498908 100644
--- i/resources/remote-control.txt
+++ w/resources/remote-control.txt
@@ -30,9 +30,9 @@ Supported commands:
  U RECORD <status>
     Set status of audio recorder to <status>
  u DSP
-    Get DSP status
+    Get DSP (reciever) status
  U DSP <status>
-    Set DSP status to <status>
+    Set DSP (reciever) status to <status>
  u RDS
     Get RDS decoder to <status>.  Only functions in WFM mode.
  U RDS <status>
```

argilo, 2023-04-28:
> Perhaps that could be improved a little?

That seems like a useful improvement. If you'd like to open a pull request, go ahead.

AsciiWolf, 2023-04-29:
@doronbehar Small typo: _rec**ie**ver_ :-)

PR #1231 (2023-04-29): Improve DSP terminology in remote-control.txt
Fix #1217.
closed 2023-04-29 by argilo