gqrx-sdr/gqrx #1227: Help, macos app won't open after packing

imyoric, 2023-04-23:
Help, macos app does not open after being packaged via macos_bundle.sh
```
aroslav@iMac-Aroslav ~ % /Users/aroslav/CLionProjects/gqrxByDSActivity/Gqrx.app/Contents/MacOS/gqrx ; exit;
dyld[35258]: Symbol not found: __ZN5boost10filesystem4pathdVERKS1_
  Referenced from: <AD88F423-9FDA-35D7-80D9-2F5C06F5F823> /Users/aroslav/CLionProjects/gqrxByDSActivity/Gqrx.app/Contents/libs/libgnuradio-fft.3.8.5.0.dylib
  Expected in:     <E292A764-4926-33BD-B438-FCFB244A96E4> /Users/aroslav/CLionProjects/gqrxByDSActivity/Gqrx.app/Contents/libs/libboost_filesystem-mt.dylib
zsh: abort      /Users/aroslav/CLionProjects/gqrxByDSActivity/Gqrx.app/Contents/MacOS/gqrx

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

```

argilo, 2023-04-23:
`macos_bundle.sh` is only used by the automated build system. I would suggest that you use the DMG releases.

imyoric, 2023-04-23:
> `macos_bundle.sh` is only used by the automated build system. I would suggest that you use the DMG releases.

I made a fork and also a couple of changes to it, how can I build it into an .app?

argilo, 2023-04-23:
You could follow the same steps that the automated build system uses: https://github.com/gqrx-sdr/gqrx/blob/master/.github/workflows/build.yml#L119-L243

imyoric, 2023-04-23:
> Вы можете выполнить те же шаги, которые использует автоматизированная система сборки: [https://github.com/gqrx-sdr/gqrx/blob/master/.github/workflows/build.yml#L119-L243 .](https://github.com/gqrx-sdr/gqrx/blob/master/.github/workflows/build.yml#L119-L243)

Thanks!!! You saved me! I've been struggling with these errors all day!

imyoric, 2023-04-24:
> Вы можете использовать те же шаги, которые используют действующую систему: [https://github.com/gqrx-sdr/gqrx/blob/master/.github/workflows/build.yml#L119-L243.](https://github.com/gqrx-sdr/gqrx/blob/master/.github/workflows/build.yml#L119-L243)

Sorry, but there is another question) Is it necessary to have a developer account?

imyoric, 2023-04-24:
> You could follow the same steps that the automated build system uses: https://github.com/gqrx-sdr/gqrx/blob/master/.github/workflows/build.yml#L119-L243

Sorry, but there is another question) Is it necessary to have a developer account?

closed 2023-04-28 by argilo