gqrx-sdr/gqrx #1229: SSB demodulator too sensitive since 2.15.10

AsciiWolf, 2023-04-27:
The SSB (and to a lesser extent also AM) demodulator seems to be too sensitive since Gqrx 2.15.10 and there is now too much clicking /  crackling in the demodulated audio.

[Here](https://github.com/gqrx-sdr/gqrx/files/11343746/beacon_recording_new.zip) is an example recording of our local beacon received with Gqrx 2.15.10.

[Here](https://github.com/gqrx-sdr/gqrx/files/11343787/beacon_recording_old.zip) is an example recording of our local beacon received with Gqrx 2.15.9.

Gain in both cases was set to -10.0 dB. Lowering the gain more has a slight effect, however there is still audible clicking / crackling even with -20.0 dB when using Gqrx 2.15.10.

It sounds even worse with gain set to 0.0 dB:

[Example recording](https://github.com/gqrx-sdr/gqrx/files/11343857/beacon_recording_new_2.zip) with Gqrx 2.15.10.

[Example recording](https://github.com/gqrx-sdr/gqrx/files/11343913/beacon_recording_old_2.zip) with Gqrx 2.15.9.

vladisslav2011, 2023-04-27:
Hello.
Could you revert this commit https://github.com/gqrx-sdr/gqrx/commit/c8c1e3f1e81d0f564e317234a6f70d4b296c6f3f and recheck?

AsciiWolf, 2023-04-27:
Thanks! I can confirm that reverting the commit did solve the issue. Here is also a screenshot from Audacity:

[image]

On the left is latest master with https://github.com/gqrx-sdr/gqrx/commit/c8c1e3f1e81d0f564e317234a6f70d4b296c6f3f reverted, on the right is latest master without any changes.

vladisslav2011, 2023-04-27:
There is alternate AGC implementation, that solves audio clipping issue independent of demodulator (NFM/WFM may have this issue too): https://github.com/gqrx-sdr/gqrx/pull/1096
The PR sequence needs to be updated with at least these commits: https://github.com/vladisslav2011/gqrx/commit/dc84c413e83e77c7ff53162376a58c5a888a0636 https://github.com/vladisslav2011/gqrx/commit/ccfe3c2aaeed7c73331c9fa712440af14f34c43b https://github.com/vladisslav2011/gqrx/commit/1e84eed1d90a9e467b1874b78f71407849c65e63 https://github.com/vladisslav2011/gqrx/commit/18e56672b3b3ff0bc5dfebc5ba0be000f4f5f7ae
You may try to rebase the PR seqence on top of current master, cherry-pick given commits and test...

AsciiWolf, 2023-04-27:
I am sadly overwhelmed by other work tasks at the moment. I may try working on it later, however I think that for now it would be the best to just revert https://github.com/gqrx-sdr/gqrx/commit/c8c1e3f1e81d0f564e317234a6f70d4b296c6f3f and maybe also release a new bugfix version of Gqrx.

/cc @argilo

vladisslav2011, 2023-04-27:
Reverting c8c1e3f1e81d0f564e317234a6f70d4b296c6f3f would bring back https://github.com/gqrx-sdr/gqrx/issues/1055 , so I don't think that it is proper solution...

argilo, 2023-04-28:
The change made in #1151 was intended to reduce CPU utilization while leaving the AGC output unchanged. It's possible that a bug was introduced. I'll see if I can spot anything.

argilo, 2023-04-28:
Aha, I think I spotted the bug: `m_DelaySamples` and `m_WindowSamples` are calculated from the old `m_SampleRate` before it is updated:

https://github.com/gqrx-sdr/gqrx/pull/1151/files#diff-04098d6b048bd5018fd5714016059258c98d0ab34f99aa6f3908abe1754d9508R142

@AsciiWolf Could you try replacing `m_SampleRate` with `SampleRate` on lines 142 and 143 to see if that resolves the issue? Thanks!

argilo, 2023-04-28:
Another (potentially better) fix would be to move lines 142 & 143 after line 148.

AsciiWolf, 2023-04-28:
I can confirm that moving lines 142 and 143 after line 148 did solve the issue:

[image]

argilo, 2023-04-28:
Thanks for confirming! I'll get a fix in shortly.

argilo, 2023-04-28:
Fix: #1230.

PR #1230 (2023-04-28): Calculate AGC parameters based on the correct sample rate
Fixes #1229.

The AGC change in #1151 introduced a bug because `m_DelaySamples` and `m_WindowSamples` were calculated prior to `m_SampleRate` being updated. This meant that the old sample rate was incorrectly used when calculating these parameters. Here I've put the calculations in the correct order.
closed 2023-04-28 by argilo