gqrx-sdr/gqrx #1234: IQ file playback fails on restart

drmpeg, 2023-04-30:
IQ file playback always fails when started, stopped and then restarted with the Start/Stop DSP button. If stop is pressed again, the whole process locks up and needs to be killed.

Parameters are:

file=/media/re/ssd/ssb.cfile,freq=100e6,rate=1.92e6,repeat=true,throttle=true

The test IQ file is here:

https://www.w6rz.net/ssb.cfile

771,446,112 bytes

drmpeg, 2023-04-30:
Built against gnuradio v3.11.0.0git-402-g54d199a2

vladisslav2011, 2023-04-30:
Could you revert https://github.com/gqrx-sdr/gqrx/commit/5052dbc060076a5b3883ab823ec3fd2e327a42dd and retest?

drmpeg, 2023-04-30:
@vladisslav2011 It's been broken for at least a year. Long before that commit.

vladisslav2011, 2023-04-30:
I can't confirm this on GNU Radio 3.7.
So it may be GNU Radio or gr-osmosdr regression.

vladisslav2011, 2023-05-01:
GNU Radio maint-3.8 branch: not confirmed
GNU Radio maint-3.9 branch: not confirmed

I have not downloaded your test file and using locally generated IQ files (~1 minute of FM band, received with RTL-SDR) to test, so the issue may be specific to your file.
Is it reproducible with any other file, recorded with Gqrx IQ Tool?

vladisslav2011, 2023-05-01:
I've reproduced something, that may match the description:
https://www.youtube.com/watch?v=gUtThjXUVOk

GNU Radio maint-3.10 branch: confirmed partially

The DSP does not freeze completely, but there is noticeable startup delay, that grows with each restart.
Another GNU Radio regression...

drmpeg, 2023-05-01:
Yes, that's the behavior. Except my IQ file is at a much lower sample rate (1.92 Msps), so it takes much longer to recover.

Testing now with 3.9-maint to confirm.

drmpeg, 2023-05-01:
Confirmed that GNU Radio maint-3.9 works fine.

I watched your video again and realized the sample rates are not that much different. I think the long recovery I'm seeing (20 seconds for the first time) is due to the size of my test file (50 seconds long).

argilo, 2023-05-01:
The problem appears to be the removal of this line of code:

 https://github.com/gnuradio/gnuradio/pull/5670/files#diff-5dbb6299f5c5a8cc7772c76182a47d525c8af7eabaef4239ab43d8a569795ebdL47

I think that should have been replaced by `d_total_items = 0;`

argilo, 2023-05-01:
Fix: https://github.com/gnuradio/gnuradio/pull/6647

drmpeg, 2023-05-03:
Fixed with latest GNU Radio commit https://github.com/gnuradio/gnuradio/commit/b7a0ac5f3abe4be258600b8367029681cbedf599

closed 2023-05-03 by drmpeg