gqrx-sdr/gqrx #1235: Freq zoom slider behaves erratically

argilo, 2023-05-04:
Sometimes the "Freq zoom" slider refuses to decrease below a certain value, or becomes completely stuck.

Scenario 1:

1. Click "Reset" in the "FFT Settings" panel.
2. Hover over the frequency axis close to the upper frequency and zoom in to ~30x using the mouse wheel.
3. Try to drag the "Freq zoom" slider back down to 1x. It will stop well before, refusing to go any lower.

Scenario 2:

1. Click "Reset" in the "FFT Settings" panel.
2. Hover over the frequency axis close to the upper frequency and zoom in to ~30x using the mouse wheel.
3. Drag the frequency axis far enough to move beyond the end of the FFT, so that part of the display is empty.
4. Try to drag the "Freq zoom" slider back down to 1x. It will jump up to a larger value and become completely stuck.

It remains possible to zoom out by other means, just not using the "Freq zoom" slider.

willcode, 2023-05-14:
How should the slider work when it represents zooms of 1-10000+? Is it the right GUI element now?

argilo, 2023-05-14:
I expect it would work much better if the scale was logarithmic instead of linear.

argilo, 2023-06-17:
I suspect scenario one has to do with zoom being centred around the demod frequency (indicated by the vertical red line in the plot). When the red line is invisible because it's outside the displayed frequency range, zooming out stops working properly.

vladisslav2011, 2023-06-18:
I think, zoom slider and wheel zoom may behave erratically due to different zoom in/ zoom out scale factors.
https://github.com/vladisslav2011/gqrx/commit/b3eeb3258b11edb027b6719b02a699d35fb97674
(Old plotter, but wheel handling code is the same)

PR #1248 (2023-08-02): dockfft: zoom slider uses log scale with max 100000x
A log scale is more sensible for the zoom slider, and it was difficult to make it work at all with a linear scale.

Fixes #1235 

Also:
- The slider is now log scaled so zoom per tick is consistent over the entire range
- Center frequency stays the same when zooming using the slider - it previously was centered on the demod freq which led to strange behavior when both mouse-based and slider-based zoom were used
closed 2023-08-02 by argilo