gqrx-sdr/gqrx #1246: Program falsely detects crash after reboot as well as SIGHUP/SIGKILL/SIGTERM.

xgpt, 2023-05-18:
I would appreciate a switch to disable the crash detected pop up, this gets in the way of some automation software I'm writing for a disabled user who needs to be able to recover after a power outage or normal reboot without someone clicking the 'crash detected' window.

He cannot see the window and is probably going to do poorly with the introduction of a screen reader program so those aren't options either.

Regardless, I believe this software should be able to start up cleanly after a reboot while running. Is there a way to correct this behavior? Or can someone guide me on how to correctly close the program without using the GUI from CLI in linux?

vladisslav2011, 2023-05-18:
Hello.
Are you using X11?
If so, `xte "keydown Control_L" "key Q" "keyup Control_L"` from a background shell may close foreground Gqrx window correctly without leaving "dirty shutdown" flag in a config file.

xgpt, 2023-05-18:
It appears as though "xte" is not installed by default on my arch linux installation (arch has gqrx 2.16, Debian is stuck on 2.15.something for now)

Although I think I'm just going to script copying a known-good config over top of the last default config to remove the crash flag! I didn't realize that would be so simple to remove the crash flag.

It would still be good for the program to handle a reboot without thinking it crashed though, it should be able to accept those signals gracefully without thinking it crashed. like if I start it from cli and then ctrl-c it, that's not a crash...

xgpt, 2023-05-18:
Right now I'm going to just
```bash
echo "[General]" >> /home/username/.config/gqrx/default.conf
echo "configversion=3" >>  /home/username/.config/gqrx/default.conf
echo "crashed=false" >>  /home/username/.config/gqrx/default.conf
```
to brute force a fix for myself...but I feel like sighup/sigterm/ a regular reboot should not generate a crash...

Ideas?

I'd prefer to stay with mainline code and not run a custom branch that has unique shutdown remote commands, would you all be open to a remote control shutdown command? What command would you like? "Q PROGRAM" or something?

xgpt, 2023-05-18:
I would recommend

```

0x87 set_powerstat 0 

```

Per https://www.mankier.com/1/rigctl to maintain compatibility with rigctld

argilo, 2023-06-18:
It looks like it's possible (although not simple) to handle Unix signals in a Qt application: https://doc.qt.io/qt-6/unix-signals.html

However, I'd prefer to avoid platform-specific code like that if possible.

The `set_powerstat` idea sounds reasonable.