gqrx-sdr/gqrx #1257: Build fail since last commit

AsciiWolf, 2023-06-18:
It looks like that Gqrx has issues building since 46031671498c834f0637c9fd880e5d5eb2a7f4b5.

```
[ 89%] Building CXX object src/CMakeFiles/gqrx.dir/receivers/nbrx.cpp.o
/__w/gqrx/gqrx/src/qtgui/plotter.cpp: In member function 'void CPlotter::zoomStepX(float, int)':
/__w/gqrx/gqrx/src/qtgui/plotter.cpp:910:80: error: no matching function for call to 'qBound(double, double, float)'
     double new_range = qBound(10.0, m_Span * (double)step, m_SampleFreq * 10.0f);
                                                                                ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtGui/qtguiglobal.h:43:0,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qcolor.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/QColor:1,
                 from /__w/gqrx/gqrx/src/qtgui/plotter.cpp:31:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:540:34: note: candidate: template<class T> constexpr const T& qBound(const T&, const T&, const T&)
 Q_DECL_CONSTEXPR inline const T &qBound(const T &min, const T &val, const T &max)
                                  ^~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:540:34: note:   template argument deduction/substitution failed:
/__w/gqrx/gqrx/src/qtgui/plotter.cpp:910:80: note:   deduced conflicting types for parameter 'const T' ('double' and 'float')
     double new_range = qBound(10.0, m_Span * (double)step, m_SampleFreq * 10.0f);
                                                                                ^
src/CMakeFiles/gqrx.dir/build.make:1383: recipe for target 'src/CMakeFiles/gqrx.dir/qtgui/plotter.cpp.o' failed
make[2]: *** [src/CMakeFiles/gqrx.dir/qtgui/plotter.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [src/CMakeFiles/gqrx.dir/all] Error 2
CMakeFiles/Makefile2:135: recipe for target 'src/CMakeFiles/gqrx.dir/all' failed
make: *** [all] Error 2
Makefile:129: recipe for target 'all' failed
Error: Process completed with exit code 2.
```

See this build log: https://github.com/gqrx-sdr/gqrx/actions/runs/5301843907/jobs/9596379602

/cc @argilo

LongnoseRob, 2023-06-18:
Build also failed for me on armv7 at the same line 910 in plotter.cpp

markjfine, 2023-06-18:
qt expects third term to be cast as double. Declaring the multiplier as a float fixes one problem but creates another one. Line 910 can be changed by encapsulating the third term and casting it as double, which fixes the problem:
   double new_range = qBound(10.0, m_Span * (double)step, (double)(m_SampleFreq * 10.0f));

argilo, 2023-06-18:
Fixed in 09a8de5f6bea6c692f84ee7921d0e2a1e45fd88b.

In the future, there's no need to report CI failures because I get automated emails any time a build fails.

closed 2023-06-18 by argilo