gqrx-sdr/gqrx #1268: Trouble compiling gqrx on Ubuntu 20.04 using three different versions

ssbrshei, 2023-07-31:
Hi,

Thanks for the wonderful package. But I have run into some compiling issue and cannot seem to get around it. Specifically, here is the script showing my problem:

---------
...
[ 28%] Building CXX object src/CMakeFiles/gqrx.dir/applications/gqrx/receiver.cpp.o
/usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp: In member function ‘void receiver::set_input_device(std::string)’:
/usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp:204:46: error: no matching function for call to ‘gr::top_block::disconnect(osmosdr::source::sptr&, int, fir_decim_cc_sptr&, int)’
  204 |         tb->disconnect(src, 0, input_decim, 0);
      |                                              ^
In file included from /usr/include/gnuradio/top_block.h:15,
                 from /usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp:30:
/usr/include/gnuradio/hier_block2.h:127:10: note: candidate: ‘void gr::hier_block2::disconnect(gr::basic_block_sptr)’
  127 |     void disconnect(basic_block_sptr block);
      |          ^~~~~~~~~~
/usr/include/gnuradio/hier_block2.h:127:10: note:   candidate expects 1 argument, 4 provided
/usr/include/gnuradio/hier_block2.h:137:5: note: candidate: ‘void gr::hier_block2::disconnect(gr::basic_block_sptr, int, gr::basic_block_sptr, int)’
  137 |     disconnect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |     ^~~~~~~~~~
/usr/include/gnuradio/hier_block2.h:137:33: note:   no known conversion for argument 1 from ‘osmosdr::source::sptr’ {aka ‘boost::shared_ptr<osmosdr::source>’} to ‘gr::basic_block_sptr’ {aka ‘std::shared_ptr<gr::basic_block>’}
  137 |     disconnect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |                ~~~~~~~~~~~~~~~~~^~~
/usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp:209:42: error: no matching function for call to ‘gr::top_block::disconnect(osmosdr::source::sptr&, int, iq_swap_cc_sptr&, int)’
  209 |         tb->disconnect(src, 0, iq_swap, 0);
      |                                          ^
In file included from /usr/include/gnuradio/top_block.h:15,
                 from /usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp:30:
/usr/include/gnuradio/hier_block2.h:127:10: note: candidate: ‘void gr::hier_block2::disconnect(gr::basic_block_sptr)’
  127 |     void disconnect(basic_block_sptr block);
      |          ^~~~~~~~~~
/usr/include/gnuradio/hier_block2.h:127:10: note:   candidate expects 1 argument, 4 provided
/usr/include/gnuradio/hier_block2.h:137:5: note: candidate: ‘void gr::hier_block2::disconnect(gr::basic_block_sptr, int, gr::basic_block_sptr, int)’
  137 |     disconnect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |     ^~~~~~~~~~
/usr/include/gnuradio/hier_block2.h:137:33: note:   no known conversion for argument 1 from ‘osmosdr::source::sptr’ {aka ‘boost::shared_ptr<osmosdr::source>’} to ‘gr::basic_block_sptr’ {aka ‘std::shared_ptr<gr::basic_block>’}
  137 |     disconnect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |                ~~~~~~~~~~~~~~~~~^~~
/usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp:240:43: error: no matching function for call to ‘gr::top_block::connect(osmosdr::source::sptr&, int, fir_decim_cc_sptr&, int)’
  240 |         tb->connect(src, 0, input_decim, 0);
      |                                           ^
In file included from /usr/include/gnuradio/top_block.h:15,
                 from /usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp:30:
/usr/include/gnuradio/hier_block2.h:83:10: note: candidate: ‘void gr::hier_block2::connect(gr::basic_block_sptr)’
   83 |     void connect(basic_block_sptr block);
      |          ^~~~~~~
/usr/include/gnuradio/hier_block2.h:83:10: note:   candidate expects 1 argument, 4 provided
/usr/include/gnuradio/hier_block2.h:93:10: note: candidate: ‘void gr::hier_block2::connect(gr::basic_block_sptr, int, gr::basic_block_sptr, int)’
   93 |     void connect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |          ^~~~~~~
/usr/include/gnuradio/hier_block2.h:93:35: note:   no known conversion for argument 1 from ‘osmosdr::source::sptr’ {aka ‘boost::shared_ptr<osmosdr::source>’} to ‘gr::basic_block_sptr’ {aka ‘std::shared_ptr<gr::basic_block>’}
   93 |     void connect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |                  ~~~~~~~~~~~~~~~~~^~~
/usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp:245:39: error: no matching function for call to ‘gr::top_block::connect(osmosdr::source::sptr&, int, iq_swap_cc_sptr&, int)’
  245 |         tb->connect(src, 0, iq_swap, 0);
      |                                       ^
In file included from /usr/include/gnuradio/top_block.h:15,
                 from /usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp:30:
/usr/include/gnuradio/hier_block2.h:83:10: note: candidate: ‘void gr::hier_block2::connect(gr::basic_block_sptr)’
   83 |     void connect(basic_block_sptr block);
      |          ^~~~~~~
/usr/include/gnuradio/hier_block2.h:83:10: note:   candidate expects 1 argument, 4 provided
/usr/include/gnuradio/hier_block2.h:93:10: note: candidate: ‘void gr::hier_block2::connect(gr::basic_block_sptr, int, gr::basic_block_sptr, int)’
   93 |     void connect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |          ^~~~~~~
/usr/include/gnuradio/hier_block2.h:93:35: note:   no known conversion for argument 1 from ‘osmosdr::source::sptr’ {aka ‘boost::shared_ptr<osmosdr::source>’} to ‘gr::basic_block_sptr’ {aka ‘std::shared_ptr<gr::basic_block>’}
   93 |     void connect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
      |                  ~~~~~~~~~~~~~~~~~^~~
/usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp: In member function ‘unsigned int receiver::set_input_decim(unsigned int)’:
/usr/local/src/soicroot/gqrx-sdr/gqrx/src/applications/gqrx/receiver.cpp:385:46: error: no matching function for call to ‘gr::top_block::disconnect(osmosdr::source::sptr&, int, fir_decim_cc_sptr&, int)’
[... 204 lines cut ...]
libgnuradio-filter3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-fosphor3.8.0/focal,now 3.8~2.2d4fe78-1build3 amd64 [installed,automatic]
libgnuradio-iio3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-iqbalance3.8.0/focal,now 0.38-4build3 amd64 [installed,automatic]
libgnuradio-network3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-osmosdr0.2.0/focal,now 0.2.0-2 amd64 [installed,automatic]
libgnuradio-pdu3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-pmt3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-pmt3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-qtgui3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-runtime3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-runtime3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-soapy3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-trellis3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-uhd3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-uhd3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-video-sdl3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-vocoder3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-wavelet3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]
libgnuradio-zeromq3.10.7/focal,now 3.10.7.0-0~gnuradio~focal-2 amd64 [installed,automatic]

Thanks in advance for your help. Sorry if this problem has been reported/resolved before. I am new to this group and I have tried my best to search for the past postings and could find a similar report.

Sincerely,
Sha

ssbrshei, 2023-07-31:
Forgot to say that I have tried: 2.14.5, 2.16, and the git cloned version. Thanks.  --Sha

argilo, 2023-07-31:
It looks like you have two different versions of GNU Radio installed (3.8.1, 3.10.7). I would suggest uninstalling one of them before attempting to build Gqrx.

ssbrshei, 2023-07-31:
Another problem is that before I tried to compile the package from source, I actually installed the package from Ubuntu 20.04 official repository.  But when ran it, it just pop a window with error:

Crash detected
gqrx exited with an exception:
logger_set_level:Bad level type.

I cannot seem to find any solution doing Google/Bing search.

ssbrshei, 2023-07-31:
> It looks like you have two different versions of GNU Radio installed (3.8.1, 3.10.7). I would suggest uninstalling one of them before attempting to build Gqrx.

Thanks for the quick reply. I will try to remove one version and try again.

ssbrshei, 2023-07-31:
> It looks like you have two different versions of GNU Radio installed (3.8.1, 3.10.7). I would suggest uninstalling one of them before attempting to build Gqrx.

Hmm... the official gnuradio and gnuradio-dev is version 3.10.7 (and I believe the latter is required for compiling gqrx-sdr. The 3.8.1 versions were pulled in when I installed the official gqrx-sdr from Ubuntu as dependencies.

argilo, 2023-07-31:
> logger_set_level:Bad level type

That sounds like another symptom of having multiple, conflicting versions of GNU Radio or out-of-tree modules on your system. I'd suggest sticking to one installation method (native packages, PPA, or building from source) rather than combining multiple.

ssbrshei, 2023-07-31:
> > logger_set_level:Bad level type
> 
> That sounds like another symptom of having multiple, conflicting versions of GNU Radio or out-of-tree modules on your system. I'd suggest sticking to one installation method (native packages, PPA, or building from source) rather than combining multiple.

Thanks for the suggestion. I have removed all 3.10.7 version:

$ apt list --installed |& grep gnuradio 
libgnuradio-analog3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-audio3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-blocks3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-digital3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-fcdproplus3.8.0/focal,now 3.8~20190817-3build3 amd64 [installed,automatic]
libgnuradio-fft3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-filter3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-fosphor3.8.0/focal,now 3.8~2.2d4fe78-1build3 amd64 [installed,auto-removable]
libgnuradio-iqbalance3.8.0/focal,now 0.38-4build3 amd64 [installed,automatic]
libgnuradio-osmosdr0.2.0/focal,now 0.2.0-2 amd64 [installed,automatic]
libgnuradio-pmt3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-runtime3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-uhd3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]

Still got the same problem:

$ gqrx
Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0

and then it poped up a window with the error I gave earlier. Any other suggestion? Thanka again.

P.S.: All the packages were installed from Ubuntu 20.04 official repository. I have no idea why they officially supoort one verion (3.10.7) for gnuradio and gnuradio-dev, for example. But then the gqrx package pulled in version 3.8.10 packages.

argilo, 2023-07-31:
I would guess that there are still things left over from whatever installation method you used to get 3.10.7.

This appears to be an installation error rather than a bug in Gqrx, so I'll close off this issue. If you'd like help with installation, I'd suggest asking in the Google group: https://groups.google.com/g/gqrx

closed 2023-07-31 by argilo