gqrx-sdr/gqrx #1272: Add support for the RTL-SDR Blog V4 to AppImage and dmg packages

rtlsdrblog, 2023-08-19:
Hi, I've just released the RTL-SDR Blog V4 (https://www.rtl-sdr.com/rtl-sdr-blog-v4-dongle-initial-release/) which is based on the R828D tuner chip. But the hardware is slightly different as it uses a different LO frequency for the R828D compared to what is hardcoded in the drivers, and the RF inputs need to be switched in a different way.

I've created a fork of the Osmocom drivers here https://github.com/rtlsdrblog/rtl-sdr-blog-minimal, and a more heavily modified one here with other unrelated changes https://github.com/rtlsdrblog/rtl-sdr-blog. These repo's were modified to add support for the V4, and the -minimal fork has the minimal amount of changed to make it work. I've submitted a patch to Osmocom to try and upstream the main changes for the V4, but the Osmocom rtl-sdr project might be dead as I see that patch requests have been unanswered for a few years. There are no changes that break compatibility with other RTL-SDR dongles.

If it's not upstreamed is there some other way to support the RTL-SDR Blog V4?

On Windows it's easy as I can just tell users to replace the .dll file, and if installed natively on Linux users just need to update the system RTL-SDR driver. But for solutions that prepackage the rtlsdr driver together with the software like AppImage and dmg for Mac's there is a problem. 

I see at https://github.com/gqrx-sdr/gqrx/blob/master/.github/workflows/build.yml#L119 you install librtlsdr from apt and brew. But perhaps you could install and compile from the https://github.com/rtlsdrblog/rtl-sdr-blog-minimal fork instead? Or the osmocom GitHub if it is indeed ever upstreamed?

argilo, 2023-08-22:
I see that the maintainer has responded on the osmocom mailing list, so it's likely your changes will be merged in soon. I'll try to pick them up in Gqrx builds soon after that happens.

rtlsdrblog, 2023-08-22:
Thanks for keeping an eye out. Yep they responded now so hopefully it will be upstreamed soon.

Anyone know whose in charge of the Macports/Homebrew rtl-sdr packages? I guess you will still need to change to build from source?

argilo, 2023-08-22:
The Macports maintainer is @michaelld: https://ports.macports.org/port/rtl-sdr/details/

The Homebrew formula appears to be community-maintained, with many contributors listed in the commit history: https://github.com/Homebrew/homebrew-core/commits/442f9cc511ce6dfe75b96b2c83749d90dde914d2/Formula/librtlsdr.rb?browsing_rename_history=true&new_path=Formula/lib/librtlsdr.rb&original_branch=master

Homebrew tends to pick up tagged releases quickly, so you might want to suggest that the rtl-sdr maintainer tag a release once your change is merged. Homebrew also accepts pull request, so you could propose your own update if needed.

If it becomes necessary, I could always build rtl-sdr from source in the binary releases of Gqrx. Some other dependencies like gr-osmosdr are already built from source.

rtlsdrblog, 2023-08-28:
Just confirming that Osmocom have upstreamed the changes now, so in your next release please do update the drivers.

PR #1280 (2023-09-20): Add support for RTL-SDR Blog V4
Closes #1272.

Here I've added support for RTL-SDR Blog V4 dongles. On Linux, the `master` branch of rtl-sdr is built from source, and on macOS the `HEAD` version is built using Homebrew. Because @rtlsdrblog's gr-osmosdr patch has not yet been accepted upstream, I've temporarily forked gr-osmosdr and applied the patch.

Pre-release binaries (AppImage and DMG) can be downloaded in the "Artifacts" section here: https://github.com/gqrx-sdr/gqrx/actions/runs/6244124081

@rtlsdrblog I'd appreciate your feedback on the pre-release versions.
closed 2023-09-20 by argilo