gqrx-sdr/gqrx #1276: Request: create a gqrx-sdr Flatpak

apsinthium, 2023-09-06:
For those like me, on Debian, packages can become fairly outdated — which is to be expected on Debian. However, for apps like gqrx-sdr, it'd be nice to have an up-to-date release in the form of a Flatpak, if at all possible.

argilo, 2023-09-19:
I would suggest using the existing AppImage releases.

I don't have any experience with Flatpak and don't have time to dive into it, so support would need to be contributed by someone else.

AsciiWolf, 2025-07-27:
Fedora and possibly other distros are considering to move to a Flatpak-only format with immutable core, and many other distributions will probably follow. And while AppImage is great and works fine, Flatpak would be much better choice for such distributions.

I will try to start working on it as soon as time allows, but can't promise anything at this point.

AsciiWolf, 2025-09-07:
I have created a Gqrx Flatpak and am currently submitting it to [Flathub](https://flathub.org).

@argilo Please let us know if you are interested in push rights to the Flatpak build manifest GitHub repository once the Flatpak is accepted and added to Flathub and possibly also in [verifying](https://docs.flathub.org/docs/for-app-authors/verification) the Flatpak.

Thanks!

closed 2026-01-03 by argilo