gqrx-sdr/gqrx #1277: Unknown remote command: ("\\get_powerstat") , when connecting from hamlib (rigctl) or wsjtx

baryluk, 2023-09-12:
I am trying to setup gqrx with wsjtx, on Debian linux, both latest stable versions, and I am getting errors on connection, because it looks like hamlib (which wsjtx uses too) by default sends some commands on connection startup. This is `\get_powerstat`, but gqrx does not understand it.

```
$ gqrx
...
Unknown remote command: ("\\get_powerstat")
```

```
strace rigctl --model=2  --rig-file=127.0.0.1:7356
...
...
socket(AF_INET, SOCK_STREAM, IPPROTO_TCP) = 5
connect(5, {sa_family=AF_INET, sin_port=htons(7356), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
getsockname(5, {sa_family=AF_INET, sin_port=htons(57252), sin_addr=inet_addr("127.0.0.1")}, [16]) = 0
ioctl(5, FIONREAD, [0])                 = 0
write(5, "\\get_powerstat\n", 15)       = 15
pselect6(6, [5], NULL, [5], {tv_sec=10, tv_nsec=0}, NULL) = 1 (in [5], left {tv_sec=9, tv_nsec=988072431})
read(5, "R", 1)                         = 1
pselect6(6, [5], NULL, [5], {tv_sec=10, tv_nsec=0}, NULL) = 1 (in [5], left {tv_sec=9, tv_nsec=999998330})
read(5, "P", 1)                         = 1
pselect6(6, [5], NULL, [5], {tv_sec=10, tv_nsec=0}, NULL) = 1 (in [5], left {tv_sec=9, tv_nsec=999999050})
read(5, "R", 1)                         = 1
pselect6(6, [5], NULL, [5], {tv_sec=10, tv_nsec=0}, NULL) = 1 (in [5], left {tv_sec=9, tv_nsec=999999040})
read(5, "T", 1)                         = 1
pselect6(6, [5], NULL, [5], {tv_sec=10, tv_nsec=0}, NULL) = 1 (in [5], left {tv_sec=9, tv_nsec=999998970})
read(5, " ", 1)                         = 1
pselect6(6, [5], NULL, [5], {tv_sec=10, tv_nsec=0}, NULL) = 1 (in [5], left {tv_sec=9, tv_nsec=999999040})
read(5, "1", 1)                         = 1
pselect6(6, [5], NULL, [5], {tv_sec=10, tv_nsec=0}, NULL) = 1 (in [5], left {tv_sec=9, tv_nsec=999999120})
read(5, "\n", 1)                        = 1
close(3)                                = 0
close(4)                                = 0
write(2, "rig_open: error = 127.0.0.1:7356"..., 60rig_open: error = 127.0.0.1:7356 No such file or directory 
) = 60
```

`RPRT 1`, which indicates a failure.

Responding letter by letter is not very efficient either, but that is a separate issue.

baryluk, 2023-09-12:
This is somehow related, and worth reading on hamlib side: https://github.com/Hamlib/Hamlib/issues/1266

argilo, 2023-09-13:
Gqrx already has already implemented a number of commands that are only needed for hamlib compatibility, so I suppose it wouldn't hurt to add more.

On my system, I also see that `\chk_vfo` is sent and responded to with `RPRT 1`, but WSJT-X seems happy to carry on after that.

kukabu, 2023-10-02:
Gqrx has 'u/U DSP' commands and I think it's possible to add get_powerstat/set_powerstat as aliases.

argilo, 2023-10-02:
The long command names for `u` and `U` are `get_func` and `set_func` respectively:

https://hamlib.sourceforge.net/html/rigctl.1.html

`get_powerstat` and `set_powerstat` have `0x88` and `0x87` as single-character forms.

PR #1334 (2024-02-09): Reply to \get_powerstat to fix wsjtx
Fixes #1277.

Maybe better to add alias to DSP ON/OFF, but this will fix wsjtx at least.
commit 3a2d5b30 (2024-02-09): Reply to \get_powerstat to fix wsjtx

Fixes #1277
closed 2024-02-09 by argilo