gqrx-sdr/gqrx #1278: Release version 2.16.1

AsciiWolf, 2023-09-18:
I think that now is an ideal time to release a new, bugfix version of Gqrx. :-)

(One more thing that could be resolved before the new release is updating librtlsdr [to support the V4 dongle](https://github.com/gqrx-sdr/gqrx/issues/1272).)

argilo, 2023-09-18:
I do have in mind to make a new release soon, but there are already enough new features in place that I think it should be 2.17.

argilo, 2023-09-29:
If you want to help, you could test the master branch and let me know if you encounter any bugs. Pre-release binaries are available here: https://github.com/gqrx-sdr/gqrx/actions?query=workflow%3ABuild+branch%3Amaster

AsciiWolf, 2023-09-29:
Thanks! I am currently not at home and have just my laptop and (non-V4) RTL-SDR dongle here, but will try to do some testing this evening. :-)

By the way, any chance [this commit](https://github.com/gqrx-sdr/gr-osmosdr/commit/636e5234f4fef8161b93b44655c6b060cd3fb6a5) could be submitted into upstream?

argilo, 2023-09-29:
> any chance this commit could be submitted into upstream?

Carl already submitted it, but it has not yet been accepted.

Unfortunately the gr-osmosdr project appears to be dead, or at least very close. All the more reason to move Gqrx over to SoapySDR. We should probably try to do this in version 2.18.

AsciiWolf, 2023-09-29:
Ah, thanks for the info!

AsciiWolf, 2023-09-29:
I have just tested (only a quick test on VHF/UHF with my RTL-SDR dongle) the new build and it seems to work fine.

argilo, 2023-10-01:
It's out!

https://github.com/gqrx-sdr/gqrx/releases/tag/v2.17

closed 2023-10-01 by argilo