gqrx-sdr/gqrx #1297: v2.17 - how to clear waterfall?

K4KDR, 2023-10-03:
Thanks for the new release!

In the past, if the '**WF Span**' time was changed, the waterfall would be **cleared**.

As of v2.17, I see that the waterfall no longer clears, so you have a partial display of two different rates until the 'old' value scrolls off the bottom of the screen.

Is there any method remaining that results in a clean waterfall?   I've toggled several of the settings but not found one.

Thanks!

-Scott,  K4KDR

willcode, 2023-10-03:
It does not clear, but the hover text at the cursor will show `[time invalid]` in areas below the dicontinuity. We'd tried both clearing and not clearing while reworking the plotter and decided not to clear. An explicit `Clear` button would be pretty simple to add, or a marker line could be drawn at the discontinuity.

K4KDR, 2023-10-03:
Thanks for replying!

A 'clear' button would be fantastic if you wouldn't mind putting that on your request list.  That would be even better than the old way of having to quickly change the span setting and then changing it back.

Thanks!

PR #1298 (2023-10-03): plotter: clear waterfall using Delete key
The waterfall is not automatically cleared after changes to the many settings that could cause it to be distorted or invalid. An explicit clear allows users who do not want to see outdated data to get rid of it.

I did not add a button, as the FFT dock is already pretty full. We can add one if the shortcut is not sufficient.

Resolves #1297
closed 2023-10-03 by argilo