gqrx-sdr/gqrx #1311: Build ok, on Ubuntu 23.10 but crah complaning of libgnuradio 3.10.5 on run

stdevPavelmc, 2023-10-10:
# Context

- OS: Ubuntu 23.10 [DEV]
- GNURadio: 3.10.7
- Gnome: 45
- Kernel: 6.5
- Cmake: 3.27.4

# Installed gnuradio packages and versions

```
pavel@laptop:~/$ aptitude search gnuradio
i   gnuradio                        - GNU Radio Software Radio Toolkit          
i   gnuradio-dev                    - GNU Software Defined Radio toolkit develop
p   gnuradio-doc                    - GNU Software Defined Radio toolkit documen
p   libgnuradio-air-modes1          - Gnuradio Mode-S/ADS-B radio - library     
i A libgnuradio-analog3.10.7        - gnuradio analog functions                 
i A libgnuradio-audio3.10.7         - gnuradio audio functions                  
i   libgnuradio-blocks3.10.7        - gnuradio blocks functions                 
i A libgnuradio-channels3.10.7      - gnuradio channels functions               
i A libgnuradio-digital3.10.7       - gnuradio digital communications functions 
i A libgnuradio-dtv3.10.7           - gnuradio digital TV signal processing bloc
i A libgnuradio-fec3.10.7           - gnuradio forward error correction support 
i A libgnuradio-fft3.10.7           - gnuradio fast Fourier transform functions 
i A libgnuradio-filter3.10.7        - gnuradio filter functions                 
i A libgnuradio-fosphor3.9.0        - gnuradio fosphor block (GPU spectrum displ
i A libgnuradio-funcube3.10.0       - Funcube Dongle controller for GNU Radio (r
p   libgnuradio-hpsdr1.0.0          - gnuradio library for OpenHPSDR protocol 1 
i A libgnuradio-iio3.10.7           - IIO blocks for GnuRadio                   
i A libgnuradio-iqbalance3.9.0      - GNU Radio Blind IQ imbalance estimator and
p   libgnuradio-limesdr3.0.2        - LimeSDR blocks for GnuRadio (runtime libra
i A libgnuradio-network3.10.7       - gnuradio network library                  
i   libgnuradio-osmosdr0.2.0        - Gnuradio blocks from the OsmoSDR project -
i A libgnuradio-pdu3.10.7           - gnuradio streaming protocol library       
i A libgnuradio-pmt3.10.7           - gnuradio pmt container library            
i A libgnuradio-qtgui3.10.7         - gnuradio Qt graphical user interface funct
p   libgnuradio-radar3.10.0         - GNU Radio Radar Toolbox - library         
p   libgnuradio-rds1                - GNU Radio FM RDS/TMC Transceiver (library)
i A libgnuradio-runtime3.10.7       - gnuradio core runtime                     
p   libgnuradio-satellites4.4.0     - library of satellite telemetry signal deco
i A libgnuradio-soapy3.10.7         - gnuradio SoapySDR support                 
i A libgnuradio-trellis3.10.7       - gnuradio trellis modulation functions     
i A libgnuradio-uhd3.10.7           - gnuradio universal hardware driver functio
i A libgnuradio-video-sdl3.10.7     - gnuradio video functions                  
i A libgnuradio-vocoder3.10.7       - gnuradio vocoder functions                
i A libgnuradio-wavelet3.10.7       - gnuradio wavelet functions                
i A libgnuradio-zeromq3.10.7        - gnuradio zeromq functions   
```

# Build finish ok
```
[...]
[ 95%] Building CXX object src/CMakeFiles/gqrx.dir/pulseaudio/pa_sink.cc.o
[ 96%] Building CXX object src/CMakeFiles/gqrx.dir/pulseaudio/pa_source.cc.o
[ 97%] Building CXX object src/CMakeFiles/gqrx.dir/qrc_icons.cpp.o
[ 98%] Building CXX object src/CMakeFiles/gqrx.dir/qrc_textfiles.cpp.o
[100%] Linking CXX executable gqrx
[100%] Built target gqrx
pavel@laptop:~/Documentos/Software/Github/gqrx/build$
```

# Run failure:

```
pavel@laptop:~/Documentos/Software/Github/gqrx/build$ sudo make install
[sudo] contraseña para pavel: 
[  1%] Automatic MOC for target gqrx
[  1%] Built target gqrx_autogen
[100%] Built target gqrx
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/share/applications/dk.gqrx.gqrx.desktop
-- Up-to-date: /usr/local/share/metainfo/dk.gqrx.gqrx.appdata.xml
-- Up-to-date: /usr/local/share/icons/hicolor/scalable/apps/gqrx.svg
-- Installing: /usr/local/bin/gqrx
pavel@laptop:~/Documentos/Software/Github/gqrx/build$ /usr/local/bin/gqrx
gqrx: error while loading shared libraries: libgnuradio-blocks.so.3.10.5: cannot open shared object file: No such file or directory
pavel@laptop:~/Documentos/Software/Github/gqrx/build$
```

willcode, 2023-10-10:
This is an envronment problem, not a Gqrx issue. For one thing, note that the error message has a different GNU Radio version than the package listing above.

argilo, 2023-10-10:
I would guess that you still have leftovers from GNU Radio 3.10.5 on your system. I built and ran Gqrx on a clean Ubuntu 23.10 system without issue.

closed 2023-10-10 by argilo

stdevPavelmc, 2023-10-10:
> I would guess that you still have leftovers from GNU Radio 3.10.5 on your system. I built and ran Gqrx on a clean Ubuntu 23.10 system without issue.

Strange, this is a fresh system!

Thanks for the notice.