gqrx-sdr/gqrx #1316: logger_set_level: Bad level type error

briankc1, 2023-10-29:
Hello - Ever since I upgraded to 22.04 LTS (jammy) I've not been able to run any gqrx Appimages - keep getting "logger_set_level: Bad level type" errors per the attached crash notice.

Any suggestions on how to resolve this? 
Thank you,
Brian
[image]

ADolbyB, 2023-10-30:
Hello, 

I have the same exact error and terminal output using Mint 20.3 (Ubuntu 20.04 LTS) Using either the AppImage or the release in the SW Manager.

Thank You,

-- Joel

willcode, 2023-10-30:
Same on Fedora 38.

argilo, 2023-10-31:
> Also, running with the lib env vars set up for a custom GNU Radio install (e.g., PyBOMBS) causes a seg fault with both main and maint-3.10 GR versions.

I expect that's because `LD_LIBRARY_PATH` causes the system's libgnuradio-osmosdr to be selected instead of the AppImage's, and libgnuradio-osmosdr depends on various GNU Radio libraries whose versions will conflict with the ones supplied by the AppImage.

argilo, 2023-10-31:
So far I haven't been able to reproduce this.

@briankc1 @ADolbyB @willcode While the error dialog is on your screen, could you open another terminal window, execute the following command, and paste the output?

```
ldd /tmp/.mount_Gqrx-*/usr/bin/gqrx
```

Also, please send the output of `echo $LD_LIBRARY_PATH`, if any.

briankc1, 2023-10-31:
Hi Clayton,
Thanks for your reply - the output of the ldd command looks like this:
    linux-vdso.so.1 (0x00007ffed5bc0000)
        libQt5Network.so.5 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libQt5Network.so.5 (0x00007f1ec7c00000)
        libQt5Svg.so.5 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libQt5Svg.so.5 (0x00007f1ec7800000)
        libgnuradio-osmosdr.so.0.2.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-osmosdr.so.0.2.0 (0x000
07f1ec7400000)
        libpulse.so.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libpulse.so.0 (0x00007f1ec7000000)
        libpulse-simple.so.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libpulse-simple.so.0 (0x00007f1ec6c00000)
        libgnuradio-digital.so.3.8.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-digital.so.3.8.2 (0x000
07f1ec6800000)
        libQt5Widgets.so.5 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libQt5Widgets.so.5 (0x00007f1ec5e00000)
        libQt5Gui.so.5 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libQt5Gui.so.5 (0x00007f1ec5600000)
        libQt5Core.so.5 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libQt5Core.so.5 (0x00007f1ec4e00000)
        libgnuradio-analog.so.3.8.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-analog.so.3.8.2 (0x00007
f1ec4a00000)
        libgnuradio-filter.so.3.8.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-filter.so.3.8.2 (0x00007
f1ec4600000)
        libgnuradio-blocks.so.3.8.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-blocks.so.3.8.2 (0x00007
f1ec4000000)
        libgnuradio-fft.so.3.8.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-fft.so.3.8.2 (0x00007f1ec3c
00000)
        libgnuradio-runtime.so.3.8.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-runtime.so.3.8.2 (0x000
07f1ec3800000)
        libvolk.so.2.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libvolk.so.2.2 (0x00007f1ec3200000)
        libgnuradio-pmt.so.3.8.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-pmt.so.3.8.2 (0x00007f1ec2e
00000)
        libboost_system.so.1.65.1 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libboost_system.so.1.65.1 (0x00007f1ec
2a00000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1ec8545000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f1ec2600000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1ec7b19000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f1ec8525000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1ec2200000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f1ec8507000)
        libboost_thread.so.1.65.1 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libboost_thread.so.1.65.1 (0x00007f1ec
1e00000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f1ec8502000)
        libgnuradio-iqbalance.so.3.8.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-iqbalance.so.3.8.0 (0
x00007f1ec1a00000)
        libgnuradio-fcdproplus.so.3.8.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-fcdproplus.so.3.8.0 
(0x00007f1ec1600000)
        librtlsdr.so.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/librtlsdr.so.0 (0x00007f1ec1200000)
        libgnuradio-uhd.so.3.8.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libgnuradio-uhd.so.3.8.2 (0x00007f1ec0e
00000)
        libuhd.so.003.010.003 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libuhd.so.003.010.003 (0x00007f1ec0200000)
        libhackrf.so.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libhackrf.so.0 (0x00007f1ebfe00000)
        libbladeRF.so.2 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libbladeRF.so.2 (0x00007f1ebfa00000)
        libairspy.so.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libairspy.so.0 (0x00007f1ebf600000)
        libairspyhf.so.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libairspyhf.so.0 (0x00007f1ebf200000)
        libSoapySDR.so.0.6 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libSoapySDR.so.0.6 (0x00007f1ebee00000)
        libfreesrp.so.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libfreesrp.so.0 (0x00007f1ebea00000)
        libpulsecommon-11.1.so => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libpulsecommon-11.1.so (0x00007f1ebe60000
0)
        libdbus-1.so.3 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libdbus-1.so.3 (0x00007f1ebe200000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1ec84f7000)
        liblog4cpp.so.5 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/liblog4cpp.so.5 (0x00007f1ebde00000)
        libvolk.so.2.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libvolk.so.2.0 (0x00007f1ebd800000)
        libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f1ec7a92000)
        libpng16.so.16 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libpng16.so.16 (0x00007f1ebd400000)
        libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f1ec7731000)
        libicui18n.so.60 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libicui18n.so.60 (0x00007f1ebce00000)
        libicuuc.so.60 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libicuuc.so.60 (0x00007f1ebca00000)
        libdouble-conversion.so.1 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libdouble-conversion.so.1 (0x00007f1eb
c600000)
        libglib-2.0.so.0 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libglib-2.0.so.0 (0x00007f1ebc200000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f1ec8579000)
        libboost_filesystem.so.1.65.1 => /tmp/.mount_Gqrx-2l9kFe4/usr/bin/../lib/libboost_filesystem.so.1.65.1 (0x0
0007f1ebbe00000)
[... cut ...]
        libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007f1ec76f9000)
        libdb-5.3.so => /lib/x86_64-linux-gnu/libdb-5.3.so (0x00007f1ec3e51000)
        libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f1ec729a000)
        libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f1ec7274000)
        libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007f1ec7251000)
        libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007f1ec6e28000)

briankc1, 2023-10-31:
Clayton - I had no output when entering:
echo $LD_LIBRARY_PATH
Thanks,
Brian

willcode, 2023-10-31:
The error is coming from `libgnuradio-runtime.so.3.8.2`. Maybe we have a log level that is valid for spdlog in more recent releases, but not for log4cpp?

argilo, 2023-10-31:
Perhaps this change broke things? https://github.com/gnuradio/gnuradio/pull/6673

It changed the default log level in gnuradio-runtime.conf from `emerg` to `critical`, which GNU Radio 3.8 doesn't accept.

argilo, 2023-10-31:
That change went into GNU Radio 3.10.7.0.

@briankc1 @ADolbyB @willcode Could you check the contents of `/etc/gnuradio/conf.d/gnuradio-runtime.conf` to see whether `debug_level` is set to `critical`? If so, changing it back to `emerg` should solve the problem.

It's also possible the bad value could be in `~/.gnuradio/config.conf`.

argilo, 2023-10-31:
Also, did you folks install GNU Radio 3.10.7.0 or later from source?

argilo, 2023-10-31:
Is it reading from a `gnuradio-runtime.conf` somewhere on your system?

willcode, 2023-10-31:
No, I don't have that file anywhere reachable without additional path variables being set.

willcode, 2023-10-31:
The app is not attempting to open that file. It is opening my `~/.gnuradio/config.conf`.

argilo, 2023-10-31:
I tested with `Gqrx-2.17.3-x86_64.AppImage`.

argilo, 2023-10-31:
I imagine that upgrading the AppImage to GNU Radio 3.10 (as https://github.com/gqrx-sdr/gqrx/pull/1301 does) would solve the problem, since 3.10 doesn't throw an exception if the log level is unknown.

ADolbyB, 2023-10-31:
Hello,

Using fresh download of AppImage Gqrx-2.17.3-x86_64.AppImage

Output from command: ldd /tmp/.mount_Gqrx-*/usr/bin/gqrx
	linux-vdso.so.1 (0x00007fffba9e2000)
	libQt5Network.so.5 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libQt5Network.so.5 (0x00007f703b07f000)
	libQt5Svg.so.5 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libQt5Svg.so.5 (0x00007f703ae20000)
	libgnuradio-osmosdr.so.0.2.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-osmosdr.so.0.2.0 (0x00007f703ab29000)
	libpulse.so.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libpulse.so.0 (0x00007f703a8d9000)
	libpulse-simple.so.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libpulse-simple.so.0 (0x00007f703a6d4000)
	libgnuradio-digital.so.3.8.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-digital.so.3.8.2 (0x00007f703a380000)
	libQt5Widgets.so.5 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libQt5Widgets.so.5 (0x00007f7039ac0000)
	libQt5Gui.so.5 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libQt5Gui.so.5 (0x00007f70392f5000)
	libQt5Core.so.5 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libQt5Core.so.5 (0x00007f7038b6c000)
	libgnuradio-analog.so.3.8.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-analog.so.3.8.2 (0x00007f70388e1000)
	libgnuradio-filter.so.3.8.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-filter.so.3.8.2 (0x00007f70385f5000)
	libgnuradio-blocks.so.3.8.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-blocks.so.3.8.2 (0x00007f7038104000)
	libgnuradio-fft.so.3.8.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-fft.so.3.8.2 (0x00007f7037ec8000)
	libgnuradio-runtime.so.3.8.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-runtime.so.3.8.2 (0x00007f7037bbd000)
	libvolk.so.2.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libvolk.so.2.2 (0x00007f70376f9000)
	libgnuradio-pmt.so.3.8.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-pmt.so.3.8.2 (0x00007f7037496000)
	libboost_system.so.1.65.1 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libboost_system.so.1.65.1 (0x00007f7037290000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f703724e000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f703706a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f7036f1b000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f7036f00000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7036d0e000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f7036cf2000)
	libboost_thread.so.1.65.1 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libboost_thread.so.1.65.1 (0x00007f7036ac3000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f7036ab7000)
	libgnuradio-iqbalance.so.3.8.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-iqbalance.so.3.8.0 (0x00007f70368a1000)
	libgnuradio-fcdproplus.so.3.8.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-fcdproplus.so.3.8.0 (0x00007f7036674000)
	librtlsdr.so.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/librtlsdr.so.0 (0x00007f703645e000)
	libgnuradio-uhd.so.3.8.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libgnuradio-uhd.so.3.8.2 (0x00007f70361fe000)
	libuhd.so.003.010.003 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libuhd.so.003.010.003 (0x00007f7035665000)
	libhackrf.so.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libhackrf.so.0 (0x00007f703545c000)
	libbladeRF.so.2 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libbladeRF.so.2 (0x00007f70351c2000)
	libairspy.so.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libairspy.so.0 (0x00007f7034fb9000)
	libairspyhf.so.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libairspyhf.so.0 (0x00007f7034db2000)
	libSoapySDR.so.0.6 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libSoapySDR.so.0.6 (0x00007f7034b75000)
	libfreesrp.so.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libfreesrp.so.0 (0x00007f7034963000)
	libpulsecommon-11.1.so => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libpulsecommon-11.1.so (0x00007f70346da000)
	libdbus-1.so.3 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libdbus-1.so.3 (0x00007f7034487000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f7034481000)
	liblog4cpp.so.5 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/liblog4cpp.so.5 (0x00007f7034239000)
	libvolk.so.2.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libvolk.so.2.0 (0x00007f7033d7a000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f7033cf0000)
	libpng16.so.16 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libpng16.so.16 (0x00007f7033abc000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f70339b8000)
	libicui18n.so.60 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libicui18n.so.60 (0x00007f70334a2000)
	libicuuc.so.60 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libicuuc.so.60 (0x00007f70330c0000)
	libdouble-conversion.so.1 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libdouble-conversion.so.1 (0x00007f7032ead000)
	libglib-2.0.so.0 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libglib-2.0.so.0 (0x00007f7032b87000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f703b8c3000)
	libboost_filesystem.so.1.65.1 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libboost_filesystem.so.1.65.1 (0x00007f7032969000)
	libfftw3f.so.3 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libfftw3f.so.3 (0x00007f7032553000)
	libfftw3f_threads.so.3 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libfftw3f_threads.so.3 (0x00007f703234b000)
	libboost_program_options.so.1.65.1 => /tmp/.mount_Gqrx-2BdD3f9/usr/bin/../lib/libboost_program_options.so.1.65.1 (0x00007f70320b0000)
[... cut ...]
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f702cd89000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f702cc4a000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f702cc30000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f702cc0d000)

> So far I haven't been able to reproduce this.
>
> @briankc1 @ADolbyB @willcode While the error dialog is on your screen, could you open another terminal window, execute the following command, and paste [...]

argilo, 2023-10-31:
@ADolbyB Do you have `debug_level` / `log_level` settings in `~/.gnuradio/config.conf` or `/etc/gnuradio/conf.d/gnuradio-runtime.conf`? If so, what are the values?

briankc1, 2023-10-31:
Per attached, It works now! Changing debug_level from critical to emerg (via etc folder) did the trick.
Thanks so much everyone! :)

briankc1, 2023-10-31:
FYI - initially, I looked for the etc folder via my folder viewer (Dolphin) but couldn't find it, which I thought was odd, since I've readily found that folder in older Ubuntu versions (I'm using 22.04) ...even though I have "show hidden folders" enabled. 
However, when I typed: sudo nano /etc/gnuradio/conf.d/gnuradio-runtime.conf in the terminal, it came right up, showing the contents of the etc folder... I just looked down a short ways to see the listing for the deubug entry, edited it, saved it...closed the terminal window... then I was able to launch gqrx right away...

briankc1, 2023-10-31:
I did a clean install to obtain 22.04 - in fact, to my surprise, when I look in my application list, I don't show anything for gnuradio as a stand alone app - I'm wondering if it was installed back when I installed SDR++ or CubicSDR, both of which worked OK prior to making changes to the debug entry of gnuradio so that I could get GQRX to work. 
Also note, that after changing critical to emerg in the debug entry of gnuradio, I went back to confirm that both SDR++ and CubicSDR are still working OK....

argilo, 2023-11-01:
@briankc1 Thanks for confirming my suspicion that `debug_level = critical` is causing trouble.

I think the path forward is for Gqrx to update its AppImage releases to use GNU Radio 3.10, which doesn't throw an exception when it encounters an unknown log level in a configuration file.

ADolbyB, 2023-11-03:
> @ADolbyB Do you have `debug_level` / `log_level` settings in `~/.gnuradio/config.conf` or `/etc/gnuradio/conf.d/gnuradio-runtime.conf`? If so, what are the values?

Thank You!

I can confirm also that `/etc/gnuradio/conf.d/gnuradio-runtime.conf` had `debug_level = critical` and changing to `debug_level = emerg` worked for me.

Sorry for the late reply.

Thanks again,

Joel

argilo, 2023-11-03:
Thanks for confirming. I'm glad to hear that the workaround was successful.

I'll leave this issue open until the AppImage is updated to GNU Radio 3.10.

ss32, 2024-01-08:
Similar solution here editing `/etc/gnuradio/conf.d/gnuradio-runtime.conf` but is `emerg` a valid log level for`gnuradio`?

```
# levels, in ascending order of severity:
# trace, debug, info, warning, error, critical, off
```

argilo, 2024-01-08:
Yes it is. For backwards compatibility, `emerg` is still accepted:

https://github.com/gnuradio/gnuradio/blob/148a10913df03180e01bb63713c256683b5ad783/gnuradio-runtime/lib/logger.cc#L109-L124