gqrx-sdr/gqrx #1324: [BUG] SegFault on start

grappas, 2023-11-25:
In Arch Linux, both git and stable package, Hyprland Nightly compositor.

```
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.8.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
rx_nb_cc :info: set_min_output_buffer on block 12 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /home/grappas/.config/gqrx/bandplan.csv
BookmarksFile is /home/grappas/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 13.2.1 20230801; Boost_108300; UHD_4.5.0.0-0-unknown
ATTENTION: default value of option vblank_mode overridden by environment.
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.8.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[1]    2721868 segmentation fault (core dumped)  gqrx
```

argilo, 2023-11-25:
Could you install gdb, then run `gdb gqrx`, and when the crash occurs type `bt` and paste the output here?

grappas, 2023-11-25:
Well... now when I'm trying to address this bug I can't reproduce it...
I'm closing this issue for now.

closed 2023-11-25 by grappas