gqrx-sdr/gqrx #1327: Error compiling gqrx on Linux Mint 21.2

ON4CDJ, 2023-12-05:
Any of you guys have an idea what is going wrong here when compiling gqrx ?
libboost, gnuradio and gr-osmosdr are installed.
gqrx 2.15 did work perfectly, aswell as new AppImage.
But compiling the 2.17.3 version from source give errors, see screenshot.
Mni tnx !

Patrick

[image]

argilo, 2023-12-05:
It looks like you're trying to build against GNU Radio 3.7. Gqrx now requires 3.8 or higher.

ON4CDJ, 2023-12-05:
D'oh ! 
Ok thought it still worked with 3.7
Thanks, will install the latest GNU Radio, 3.10 that is.
So that should do the trick then I hope.

argilo, 2023-12-05:
Sounds good.

Make sure to remove 3.7, since Gqrx is likely to get confused if you have multiple GNU Radio versions installed.

ON4CDJ, 2023-12-06:
Had no time today to delete 3.7 and install the latest GNU Radio, but about sure this will be the needed solution for my error.
Many thanks Clayton !
73.

argilo, 2023-12-06:
Sounds good. If that doesn't solve the problem, let me know and we can re-open this issue.

ON4CDJ, 2023-12-07:
Unfortunately I still got these errors after deleting completely GNU Radio and installed the latest greatest thru git clone, version 3.11.0
After executing 'make' it all goes well to 100%, but then a whole bunch of errors.
Still puzzled, as I am far from a linux expert ;-)
Screenshot attached:
[image]

argilo, 2023-12-07:
Youl have to remove that and build the latest version of gr-osmosdr against your new GNU Radio.

ON4CDJ, 2023-12-07:
Ok....reinstalled gr-osmosdr, rebuilt GNU Radio, same with gqrx.....bingo !
Finally all came together as it should, and got a working gqrx v2.17.3....yay !
Thank you Clayton for pointing me in the right direction.
Now this issue really is closed ;-)
73 !
Patrick ON4CDJ

argilo, 2023-12-07:
Glad to hear it. Enjoy!