gqrx-sdr/gqrx #1328: Regressioin in waterfall

alexbirkett, 2023-12-16:
Commit 9fb0eb60bb20c73f49e7bc09bbd9fd609fe51194 seems to break the waterfall for me me:

9fb0eb60bb20c73f49e7bc09bbd9fd609fe51194 looks like this:

[image]

Prior to 9fb0eb60bb20c73f49e7bc09bbd9fd609fe51194:
[image]

  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro17,1
  Model Number:	Z11D0000ZH/A
  Chip:	Apple M1
  System Version:	macOS 14.1.1 (23B81)
  Kernel Version:	Darwin 23.1.0
  Boot Volume:	Macintosh HD

willcode, 2023-12-16:
Does the dmg in the releases area work for you?

alexbirkett, 2023-12-16:
No the problem is reproducible with `Gqrx-2.17.3.dmg`

argilo, 2023-12-16:
Can you show the contents of the FFT Settings pane?

alexbirkett, 2023-12-17:
[image]

Some of the other color maps appear to be working. It's just the GQRX color map that does not

alexbirkett, 2023-12-17:
Ok this is just user error on my part. I got caught out be some changes to the settings, I assume.

[image]

closed 2023-12-17 by alexbirkett

argilo, 2023-12-17:
What setting fixed it? Did you have the WF dB slider set too high?