gqrx-sdr/gqrx #1336: [Feature request] Basic TX support

AsciiWolf, 2024-02-12:
I know that Gq**rx** is a RX (not TX) focused SDR software, however it could also be used as a basic TX SDR as well since it already supports multiple SDRs with TX abilities. Good start could be adding some basic AM/SSB and NFM modulators.

Feel free to close this ticket if you believe that it is not a good idea. :-)

Thanks!

argilo, 2024-02-12:
I wouldn't be against adding some TX capabilities, but don't have any plans to work on it myself.

I did once build a basic transceiver using GNU Radio: https://github.com/argilo/contest-sdr

Unfortunately, what I found is that most SDRs and their associated drivers (with the exception of USRP) are built only with continuous transmit in mind; burst transmissions and T/R switching usually don't work well. Samples are often dropped from the start and/or end of a transmission, and buffered samples from the end of one transmission often reappear at the start of the next transmission. In contest-sdr I worked around that by padding transmissions with silence, but I had to manually adjust the amount of silence for each SDR, which was a pain. This also means that T/R switching is painfully slow for some SDRs.

Keeping latency under control in a GNU Radio flow graph can also be a challenge. (A possible solution: https://www.youtube.com/watch?v=jq0RewceCwc)

Most SDRs supported by gr-osmosdr / SoapySDR are also not really useful for over-the-air transmission out of the box, since they need external amplification, as well as filtering to remove harmonics.

So I certainly wouldn't place a high priority on adding TX support, but if someone wants to take a crack at it and has the patience to deal with poorly behaving hardware then go for it. If the code is maintainable and it performs well, then I would consider merging it.

priyanshumakes, 2026-11-03:
Hi @AsciiWolf, I'm considering contributing to this bounty. With #1475 and #1480 already open, are you still accepting another implementation?

I don't have SDR transmit hardware. Would you consider a contribution where I provide the code and available software-test results, with hardware validation done by you or another contributor?

I'd use AI assistance and would need to meet your code-understanding requirement before submitting.

Rohank143, 2026-11-03:
Hi @argilo and @AsciiWolf,

I have submitted a complete and robust implementation of basic TX support in PR #1483.

### Design & Architectural Highlights
1. **Continuous Flowgraph Streaming with Zero-Latency PTT Gating**:
   A major flaw with stopping and restarting the top-block on PTT events is that re-initializing hardware buffers causes severe audio pops, buffer underruns, and hardware lockups on full/half-duplex SDR devices (HackRF, PlutoSDR, LimeSDR, USRP). In PR #1483, the GNU Radio flowgraph runs continuously while the output is gated through an atomic multiplier block (`0.0f` idle, `1.0f` keyed). This yields instantaneous, click-free PTT transitions with zero buffer churn.

2. **Clean Separation & Decoupling**:
   - `src/dsp/tx_modulator`: Hierarchical GNU Radio block supporting AM, NFM (configurable deviation), USB, LSB (Hilbert analytic filter), and CW (5 ms raised-cosine envelope shaping to prevent key clicks). Includes CTCSS sub-audible tone synthesis across all 38 EIA standard tones (67.0 Hz to 250.3 Hz) and rational resampling from 48 kHz audio rate to RF baseband rate.
   - `src/applications/gqrx/transmitter`: High-level transmitter engine wrapping `osmosdr::sink` and audio input fallback.
   - `src/qtgui/docktxctl`: Modular Qt dock widget providing visual PTT button, split mode & offset controls, CTCSS tone selector, CW pitch/WPM, and RF power / microphone gain sliders.
   - `receiver.cpp`: Kept decoupled; half-duplex coordination is handled cleanly through `MainWindow`.

3. **Acoustic & Local RF Feedback Loop Elimination**:
   - Transmit activation automatically mutes receiver audio (`rx->set_audio_mute(true)`) to prevent acoustic howling loops and local receiver saturation.

4. **Hamlib rigctl Remote Control Protocol Support**:
   - `t` / `\get_ptt` & `T` / `\set_ptt`: Interrogates and toggles PTT state.
   - `s` / `\get_split_vfo` & `S` / `\set_split_vfo`: Manages split VFO state and reports TX VFO.
   - `i` / `\get_split_freq` & `I` / `\set_split_freq`: Manages transmit frequency.
   - `u PTT` & `U PTT <status>`: Queries and sets PTT function state.
   - `l RFPOWER` & `L RFPOWER <power>`: Reports and sets transmit RF power with clamp protection (0.00 to 1.00).
   - `l MICGAIN` & `L MICGAIN <gain>`: Reports and sets microphone gain (-30 to +30 dB).
   - `\dump_state`: Enhanced state dump reflecting transmit capability.

5. **Verification & AppStream Compliance**:
   - Added automated unit test suite in `tests/test_tx.cpp` verifying CTCSS frequencies, continuous PTT gating, AM/FM/CW DSP math, and remote control protocol parsing (100% test pass).
   - `dk.gqrx.gqrx.appdata.xml` updated and strictly validated with `appstreamcli validate --no-net` (0 errors).
   - `resources/remote-control.txt` updated with comprehensive documentation for all new remote commands.

PR is ready for review: https://github.com/gqrx-sdr/gqrx/pull/1483

AsciiWolf, 2026-11-03:
> Hi [@AsciiWolf](https://github.com/AsciiWolf), I'm considering contributing to this bounty. With [#1475](https://github.com/gqrx-sdr/gqrx/pull/1475) and [...]

Submissions from developers who have experience with actual SDR hardware and do not use LLMs are preferred.

mau16ro, 2026-11-05:
/attempt #1336

Proposed solution submitted for review in PR #1486 (https://github.com/gqrx-sdr/gqrx/pull/1486).

pratyushadk, 2026-11-05:
Hi,

I've been researching this feature request and am currently working on an implementation. Before going further, I wanted to share the high-level approach and get early feedback.

**What I've studied:**
I went through the existing Gqrx architecture in depth, as well as Argilo's [contest-sdr](https://github.com/argilo/contest-sdr) which was referenced in this thread. That work was helpful for understanding real-world hardware behaviour — particularly around buffering differences, TX/RX switching costs, and device-specific quirks across HackRF, USRP, PlutoSDR, LimeSDR and bladeRF.

**Proposed approach (high level):**
The core idea is to introduce TX as a fully isolated subsystem that shares the existing GNU Radio top block but does not structurally modify the receiver. The key insight driving the design is that half-duplex devices (HackRF) and full-duplex devices (USRP, Pluto, Lime, bladeRF) need fundamentally different switching strategies, and the architecture handles both. Modulation (NFM, AM, SSB) is built on existing GNU Radio blocks — no custom DSP. PTT is exposed via the GUI, a keyboard shortcut, and the existing Hamlib rigctld interface.

The goal is that a failure in the TX path cannot affect receiver stability or audio output.

**Current status:**
Actively implementing. I'll open a draft PR once the core TX chain and device handling are in a reviewable state.

A few questions for the maintainers:
- What is the minimum GNU Radio version Gqrx intends to support going forward? (This affects which sink block is most appropriate.)
- Is there a preference for how TX is exposed in the UI — as a new dock, or integrated into an existing one?
- Are there any active branches on this I should be aware of to avoid duplicate work?

Happy to discuss any of this before the PR is opened.

Thanks

---

AsciiWolf, 2026-11-05:
> What is the minimum GNU Radio version Gqrx intends to support going forward?

> Is there a preference for how TX is exposed in the UI — as a new dock, or integrated into an existing one?

These are probably questions for @argilo. :)

> Are there any active branches on this I should be aware of to avoid duplicate work?

There are some PRs already opened. #1475 is probably the most complete one.

I have one question for you: Are you an AI agent or is the development work done by actual human developer? Thanks!

pratyushadk, 2026-11-05:
Haha, I'm a human behind this! 😄 I'm using AI coding assistants to help with the development, but the research, architectural decisions and overall design direction are things I'm working through myself. I'm studying the existing implementations by @argilo, comparing different approaches and trying to understand the underlying challenges before settling on a design. AI is helping me with the coding, but I'm actively driving the engineering decisions.

hashesgenb65-dev, 2026-11-11:
Hi, I’d like to take a first step on this. Rather than touch the device or GUI code, I’d start with offline TX modulator blocks in src/dsp/, written to mirror the existing rx_demod_* blocks: an NFM one first, then SSB. They would take audio and produce I/Q, with a hard limiter so deviation is bounded, and no osmosdr::sink, PTT, or hardware path yet. Given your notes about burst TX and T/R behaviour on most drivers, I’d rather settle the DSP structure with you before any device code. Two questions: is that a sensible first PR, and would you prefer it with no test infrastructure at all, or with a small optional CMake test target? I noticed the repo has none today.

hashesgenb65-dev, 2026-11-11:
I’ve started on this and have a first building block: tx_mod_nfm, a narrow-band FM modulator (gr::hier_block2, modelled on rx_demod_fm). Chain is audio → 300 Hz–3 kHz band-pass → limiter → 3 kHz low-pass → clamp → frequency_modulator_fc, so deviation can’t exceed max_dev even when over-driven. I have an offline check (no hardware, just a vector sink) covering deviation scaling, out-of-band audio rejection, and occupied bandwidth.

Branch: https://github.com/hashesgenb65-dev/gqrx/tree/tx-nfm-modulator

It’s a standalone block and nothing uses it yet. Before I go further (AM/SSB modulators, then the flowgraph side), I’d like to know how you’d prefer TX to be structured: a separate TX top block alongside receiver, or extending receiver itself? Also, is there a preferred place for DSP tests, since Gqrx doesn’t seem to have a test directory? Happy to adapt to whatever you prefer, or to open a draft PR if that’s easier to discuss against.

AsciiWolf, 2026-11-11:
Probably a separate TX top block, but this is a question for @argilo.

Thanks!

hashesgenb65-dev, 2026-11-11:
@argilo, I’m looking at contributing to this issue and want to check direction before writing more code. I have two small DSP building blocks plus a test setup on my fork:

tx_mod_nfm and tx_mod_am (src/dsp/), gr::hier_block2 blocks modelled on rx_demod_fm/rx_demod_am. Both condition the audio (300 Hz to 3 kHz band-pass, limiter, low-pass, clamp) so deviation and envelope stay bounded when over-driven.
Offline checks under tests/, enabled with -DGQRX_BUILD_TESTS=ON and run with ctest. They need no hardware and cover deviation or depth scaling, out-of-band audio rejection, and occupied bandwidth.

Branches: https://github.com/hashesgenb65-dev/gqrx/tree/tx-am-and-tests (stacked on tx-nfm-modulator).

Questions before I go further (SSB, then the flowgraph side):

1. Do you prefer a separate TX top block alongside receiver, or extending receiver?

2. What testing do you expect on hardware (USRP, HackRF, PlutoSDR) before a TX PR could be accepted?

3. Would you rather see this as a series of small PRs (modulators, then top block, then GUI/rigctl), or one larger PR?

AsciiWolf, 2026-11-11:
The payout addresses are not valid. The only valid one is **via Stripe**, directly using BountyHub.

Also, I am not interested in more vibe coded PRs. The new ones should all be done by **actual human developers** who ideally own SDR hardware to test their changes on.

Thanks for the understanding.

Rohank143, 2026-11-12:
Hi @AsciiWolf, understood on both points:

1. **Payouts:** My submission in #1483 uses BountyHub directly (Stripe payout only — no external payout addresses).
2. **Code quality & hardware testing:** Completely agree that unreviewed "vibe-coded" PRs shouldn't be merged blindly, especially for RF transmit paths. PR #1483 was submitted earlier on Nov 3 with isolated `tx_modulator` / `transmitter` blocks, AppStream validation, and offline unit tests (`tests/test_tx.cpp`). I'm happy to walk through any part of the DSP/flowgraph/rigctl changes, adjust the architecture to match whatever direction you and @argilo prefer, and iterate on any feedback from physical SDR hardware testing.