gqrx-sdr/gqrx #1345: Freq. correction causes Segmentation fault

srs4511351, 2024-04-05:
When I set a frequency correction in gqrx 2.17.3-14-gbdb7579, it works while the application is running.
If I quit and re-run gqrx, I get a Segmentation fault when using using device string `hwVer=255,sdrplay=0`

I have a Raspberry Pi 4B with an SDRplay RSP1A.
Deleting the line corr_freq=-1000000 from default.conf lets gqrx run again.

Adjusting Freq. correction before hitting Start DSP also causes a Segmentation fault

It works with the Soapy device `driver=sdrplay,serial=19110C0797,soapy=1`.
gqrx runs and displays the frequency correction.
Switching back to `hwVer=255,sdrplay=0` with a Freq. correction in place causes an immediate Segmentation fault.

argilo, 2024-04-05:
Unfortunately I don't have an SDRplay, so I don't have any way to test this. I'll need your help with debugging. Can you run Gqrx inside gdb and grab a stack trace (`bt`) when it crashes?

Also, please let me know what operating system you are running, what version of gr-osmosdr you're using, and how you installed gr-osmosdr.

gr-osmosdr's SDRplay driver appears to ignore the frequency correction, so I'm confused why a crash would occur.

argilo, 2024-04-05:
I did a bit of searching and spotted an experimental fork of gr-osmosdr with updated SDRplay support, and it does look to pass through frequency correction to the SDRplay driver.

If you're using a fork of gr-osmosdr and the crash occurs in `set_freq_corr`, then the problem almost certainly lies in gr-osmosdr or the SDRplay driver and not Gqrx.

argilo, 2024-04-05:
Since this is very likely a bug in gr-osmosdr, I'll close this off for now. If you find otherwise, we can open this again.

closed 2024-04-05 by argilo

srs4511351, 2024-04-05:
Which fork did you find with updated SDRplay support? I need a good one. 
The osmocom gr-osmosdr does not support a recent version of the SDRplay API so I use
https://github.com/fventuri/gr-osmosdr

If that is the one you are referring to, I’ll open an issue there. 

Thanks!

argilo, 2024-04-05:
The fventuri fork is the one I was referring to.

argilo, 2024-04-05:
I see there is already an issue open since 2021: https://github.com/fventuri/gr-osmosdr/issues/2