gqrx-sdr/gqrx #1346: Enhancement: macOS Apple Silicon (arm64) builds

alexsavage, 2024-04-05:
The current build process for macOS [creates an x86_64 'thin' binary](https://github.com/gqrx-sdr/gqrx/actions/runs/8442268325/job/23129079724#step:8:1243). As a result, Apple Silicon (arm64) Macs run in [Rosetta 2 translation](https://developer.apple.com/documentation/apple-silicon/about-the-rosetta-translation-environment), which means a performance hit.

For Gqrx itself, it looks like CMake has a [CMAKE_OSX_ARCHITECTURES](https://cmake.org/cmake/help/latest/variable/CMAKE_OSX_ARCHITECTURES.html) variable to request cross-compilation on Apple platforms, including macOS. Setting this to `x86_64;arm64` would create a fat binary, instead of only targeting the build host's architecture.

One challenge will be that Homebrew (used to pull many dependencies in the `build.yml` CI workflow) [does not plan on supporting fat binaries](https://github.com/orgs/Homebrew/discussions/3706). Also, the CI steps pulling loose repos may need adjustments. But the GitLab macos-13 and macos-14 runners are arm64 hosts, so tidying up cross-compilation will probably need to be solved sooner or later.

I will play around with this a bit, but logging an issue first for feedback and discussion.

Thanks and 73!

argilo, 2024-04-05:
There's already an issue open for this, so I'll close this as a duplicate of #1104.

closed 2024-04-05 by argilo

alexsavage, 2024-04-05:
Ah thanks! I searched open issues and missed it. Much appreciated.