gqrx-sdr/gqrx #1348: Latest build breaks remote control on Mac M1

markjfine, 2024-04-19:
Am using a SwiftSocket to port 7356 via localhost (127.0.0.1) to control GQRX. Somehow this is no longer being accepted in 2.17.5, but worked in 2.17.4-14. Unless I missed something, I see nothing in the commits from the past 24 hours that could cause this. 🤷🏻‍♂️

argilo, 2024-04-19:
Processing of remote control commands was changed in #1343.

Perhaps you could record traffic on port 7356 using Wireshark to see what's going on.

argilo, 2024-04-19:
If you want to check whether that change made the difference, you could compare these two pre-release builds:

* https://github.com/gqrx-sdr/gqrx/actions/runs/8287395663 (before)
* https://github.com/gqrx-sdr/gqrx/actions/runs/8365249584 (after)

markjfine, 2024-04-19:
Looking deeper, my initial assessment that it was the latest build is false. Turns out the last time I updated and built from github commits to master was 18 Feb, and apparently the remote code was updated mid-March. I generally do a github pull weekly, so I'm not sure how I missed it the code change. Let me investigate a little more and see if I can't see what's going on and give you better feedback.

markjfine, 2024-04-19:
For whatever reason, it seems my version of Qt has a problem with QTcpSocket->canReadLine(), so the loop never gets entered. However, changing the first line in RemoteControl::startRead() from
`    while (rc_socket->canReadLine()) {`
to
`    while (rc_socket->bytesAvailable() > 2) {`
seems to work fine as an alternative.

argilo, 2024-04-19:
Which version of Qt are you using?

Can you make a packet capture with Wireshark to see what's actually happening on the wire?

argilo, 2024-04-19:
Also, which operating system are you using?

argilo, 2024-04-19:
Does your client send a newline character at the end of each command?

argilo, 2024-04-19:
Do you observe the same problem if you connect to the remote control server "by hand" with nc / telnet?

markjfine, 2024-04-19:
Will have to install wireshark and get back to you.
Qt is Qt 5.15.13
macOS is Sonoma 14.4.1 (arm)
and I believe client adds \r instead of \n at the end - which could be part of the problem.

argilo, 2024-04-19:
> Qt is Qt 5.15.13

It's probably worth updating to Qt 6, since there's at least one known bug affecting Qt 5 on modern macOS: #1073

> and I believe client adds `\r` instead of `\n` at the end

If so, that's likely the issue. `canReadLine` expects `\n`.

markjfine, 2024-04-19:
Actually... that was indeed the problem. Changed all my \r to \n, reinserted canReadLine() in the while statement and now it works just fine. #StupidLegacyMacTrix.

<headslap> So of course something named canreadline() is just going to wait until it sees a newline character. My bad, cheers.

markjfine, 2024-04-19:
I did know about the Qt5 issues, but I still have some legacy stuff that's never moved to 6 and trying to maintain both under Homebrew is real pain.

argilo, 2024-04-19:
I'm glad to hear that switching to `\n` solved it. I'm a bit surprised that the `readLine` was working, since it also reads until the first `\n`, but maybe it bails out if one never shows up.

Hopefully this change in behaviour doesn't affect any other clients. I guess time will tell.

argilo, 2024-04-19:
Closing for now, since the bug was outside Gqrx.

closed 2024-04-19 by argilo

markjfine, 2024-04-19:
Now I have to figure out why a months-old change is just hitting me now... unless I got distracted by other shiny things and simply forgot to revisit.